zephyrproject-rtos/zephyr #18732: net: mDNS issues

lbise, 2019-08-28:
I'm trying to use mDNS between two Zephyr nodes. I only want to resolve IPv6 addresses and I want to use the default server [ff02::fb]:5353.
I call dns_resolve_init with the address to setup the resolving and then send queries using the same code than samples/net/dns_resolve.
The query is properly formatted and sent.

    Frame 1955: 78 bytes on wire (624 bits), 78 bytes captured (624 bits) on interface 0
    Raw packet data
    Internet Protocol Version 6, Src: fe80::200:5eff:fe00:5303, Dst: ff02::fb
        0110 .... = Version: 6
        .... 0000 0000 .... .... .... .... .... = Traffic Class: 0x00 (DSCP: CS0, ECN: Not-ECT)
        .... .... .... 0000 0000 0000 0000 0000 = Flow Label: 0x00000
        Payload Length: 38
        Next Header: UDP (17)
        Hop Limit: 64
        Source: fe80::200:5eff:fe00:5303
        Destination: ff02::fb
        [Source SA MAC: IcannIan_00:53:03 (00:00:5e:00:53:03)]
    User Datagram Protocol, Src Port: 51196, Dst Port: 5353
        Source Port: 51196
        Destination Port: 5353
        Length: 38
        Checksum: 0xe4e6 [unverified]
        [Checksum Status: Unverified]
        [Stream index: 16]
        [Timestamps]
    Multicast Domain Name System (query)
        Transaction ID: 0x09fb
        Flags: 0x0100 Standard query
        Questions: 1
        Answer RRs: 0
        Authority RRs: 0
        Additional RRs: 0
        Queries
            zephyr.local: type AAAA, class IN, "QM" question
                Name: zephyr.local
                [Name Length: 12]
                [Label Count: 2]
                Type: AAAA (IPv6 Address) (28)
                .000 0000 0000 0001 = Class: IN (0x0001)
                0... .... .... .... = "QU" question: False

The responder properly receives the query and responds properly:

    Frame 1956: 100 bytes on wire (800 bits), 100 bytes captured (800 bits) on interface 0
    Raw packet data
    Internet Protocol Version 6, Src: fe80::200:5eff:fe00:5382, Dst: ff02::fb
        0110 .... = Version: 6
        .... 0000 0000 .... .... .... .... .... = Traffic Class: 0x00 (DSCP: CS0, ECN: Not-ECT)
        .... .... .... 0000 0000 0000 0000 0000 = Flow Label: 0x00000
        Payload Length: 60
        Next Header: UDP (17)
        Hop Limit: 255
        Source: fe80::200:5eff:fe00:5382
        Destination: ff02::fb
        [Source SA MAC: IcannIan_00:53:82 (00:00:5e:00:53:82)]
    User Datagram Protocol, Src Port: 5353, Dst Port: 5353
        Source Port: 5353
        Destination Port: 5353
        Length: 60
        Checksum: 0xeadd [unverified]
        [Checksum Status: Unverified]
        [Stream index: 17]
        [Timestamps]
    Multicast Domain Name System (response)
        Transaction ID: 0x0000
        Flags: 0x8400 Standard query response, No error
        Questions: 0
        Answer RRs: 1
        Authority RRs: 0
        Additional RRs: 0
        Answers
            zephyr.local: type AAAA, class IN, cache flush, addr fe80::200:5eff:fe00:5382
        [Unsolicited: True]

The answer is received but dropped by the resolver.

    [net/core] [DBG] net_rx: (0x20000ab8): Received pkt 0x20002720 len 100
    [net/ipv6] [DBG] net_ipv6_process_pkt: (0x20000ab8): IPv6 packet len 100 received from fe80::200:5eff:fe00:5382 to ff02::fb
    [net/conn] [DBG] net_conn_input: (0x20000ab8): Check UDP listener for pkt 0x20002720 src port 5353 dst port 5353 family 10 chksum 0xeadd len 60
    [net/conn] [DBG] net_conn_input: (0x20000ab8): No match found.
    [net/core] [DBG] processing_data: (0x20000ab8): Dropping pkt 0x20002720

This is because since the resolver sent the query using a local port it's listening on that port but the responder, correctly I might add (see RFC), responds with both ports at 5353 as expected.

jukkar, 2019-08-28:
@lbise Thanks for good analysis, saved me lot of time :+1:

jukkar, 2019-08-28:
@lbise If you have time, please test the fix proposal.

lbise, 2019-08-28:
Hello @jukkar, thank you for the speedy patch.
Now the ports are correct, however I found 2 issues.

The response is received now, however the resolver cannot find the query ID in its list of queries because when answering the mDNS responder forces the query ID to 0
ID set to 0 in response here : https://github.com/zephyrproject-rtos/zephyr/blob/ae5db66b8cc4f39aa6e0c0748d339612414420c4/subsys/net/lib/dns/mdns_responder.c#L118
Which makes resolver fail here : https://github.com/zephyrproject-rtos/zephyr/blob/ae5db66b8cc4f39aa6e0c0748d339612414420c4/subsys/net/lib/dns/resolve.c#L374

The second issue is that now if you have responder AND resolver both enabled (which was my case), the answer doesn't even reach the resolver because the packet is routed to the responder since there are 2 listeners on the same port now. Sadly I'm not sure this is easily fixable but you know better.

         Context          Iface         Flags Local               Remote
    [ 1] 0x20000538       0x20012140    6DU   [ff02::fb]:5353     [::]:0
    [ 2] 0x200005ac       0x20012140    6DU   [::]:5353   [::]:0

         Handler    Callback      Proto   Local                   Remote
    [ 1] 0x20000f70 0x0803bc7d    UDP     [ff02::fb]:5353 [::]:0
    [ 2] 0x20000fac 0x0803bc7d    UDP     [::]:5353       [::]:0

jukkar, 2019-08-29:
@lbise Fixed the "DNS id 0" issue, also the mDNS response parsing did not work properly. Please test if it works for you now. The issue with having both mDNS resolver and responder enabled at the same time, needs to be handled separately.

closed 2019-08-29 by ioannisg

jukkar, 2019-08-30:
> The second issue is that now if you have responder AND resolver both enabled (which was my case), the answer doesn't even reach the resolver because the packet is routed to the responder since there are 2 listeners on the same port now. Sadly I'm not sure this is easily fixable but you know better.

@lbise See #18797 for support to have both mDNS responser and resolver enabled at the same time.

PR #18737 (2019-08-29): Set mDNS source port to 5353
Fixes #18732
commit f4c4d120 (2019-08-29): net: mdns: Use 5353 source port when sending the query

The source UDP port in all Multicast DNS responses MUST be 5353
as described in RFC 6762 chapter 6.

Fixes #18732