zephyrproject-rtos/zephyr #18748: net: mdns: Send queries to all interfaces

lbise, 2019-08-28:
I'm using Zephyr on a node that has 2 different interfaces configured. On this node I'm using mDNS to resolve names of devices.
The issue is that currently when using dns_get_addr_info to resolve a name I have no control on which interface it will be sent to and to me it seems that in all cases it will take the default interfaces when binding because it's just binding against a port and not again an address so the default interface is selected.
See https://github.com/zephyrproject-rtos/zephyr/blob/ae5db66b8cc4f39aa6e0c0748d339612414420c4/subsys/net/lib/dns/resolve.c#L276

lbise, 2020-04-01:
Coming back to this. I implemented a workable solution but honestly I find it a bit hackish and also it only works between Zephyr nodes from my testing.

The big issue here is that you can only have a single net_context listening to a multicast address (i.e [ff02::fb]:5353). Since a net_context is locked to a single interface that means we cannot listen to a multicast address on more than one interface at the same time. This is an issue for mDNS but I suspect for other protocols also when you have more than one interface which is my case.
@jukkar suggested I investigate the use of the unicast response bit of the mDNS which you can set when sending a query. The responder is supposed then to answer to the src unicast address of the query instead of the multicast but reading RFC 6762 is merely a suggestion and in no way an obligation. This works when you are using only Zephyr nodes, which I did then it works.
I then tried to use it to resolve my linux box name but it looks like avahi just ignores the unicast response bit so the response is ignored.

I guess there is no easy solution to this issue other than properly handling multicast address listening ? Please feel free to correct me if I'm wrong.

lbise, 2020-04-02:
ping @jukkar @pfalcon

jukkar, 2020-04-02:
Ok, good to know. As you suggest, proper multicast address listening is the way to go then.

pfalcon, 2020-04-02:
> Since a net_context is locked to a single interface that means we cannot listen to a multicast address on more than one interface at the same time.

Yes, this, to my knowledge, is the current design limitation of Zephyr network stack. And I'm not sure what's the best way to deal with it. I guess I would start with a suggestion to do just that - for different protocol implementation to accept the that there may be more than 1 interface, and be ready to listen on multiple (if user requested that). I would recommend to start with that, instead of starting to patch underlyings of the stack right away.

Another issue is that a lot of networking code in Zephyr is written without awareness of multiple interfaces. And that's for obvious reason: most people don't have them. And IMHO, we first should address that. I had an idea of implementing an explicit loopback interface, so maybe we should do that finally. Then we will be able to write tests for multi-interface functionality, etc.

Not doing it in such a sustainable way will mean that we'll just add more bugs ahead of new functionality.

jukkar, 2020-04-02:
> Another issue is that a lot of networking code in Zephyr is written without awareness of multiple interfaces.

There should not be many places in core stack that does not support multiple interfaces. If one enables VLAN in Ethernet that brings in multiple interfaces immediately. I have been fixing these things when I have found them but of course there are still those left like in this mdns thingy. Please report any issues with multiple interfaces so those can be fixed. This mdns issue needs to be fixed, not sure how big task would it be.

pfalcon, 2020-04-02:
> There should not be many places in core stack that does not support multiple interfaces.

"Support" is not exactly what I mean, more of "handle properly". We used to have reports of interface not being properly selected, and @lbise's report here also "it seems that in all cases it will take the default interfaces when binding".

> I have been fixing these things

Yes, thanks.

> Please report any issues with multiple interfaces so those can be fixed.

As I mentioned in other tickets, these days (months) I'm working on setting up CI tests to run on real hardware. Keeps going much slower than desired, but if I don't miss next Zephyr networking forum, I hope to report something there. And once I finish covering basics like dumb_http_server/big_http_download, I plan to proceed with more involved testcases, then hopefully there will be reports/fixes.

closed 2023-11-17 by carlescufi

lbise, 2023-11-27:
Thanks for the fix!

PR #65247 (2023-11-17): net: mdns: Create a listener to all available network interfaces
Instead of just listening first network interface in the system for mDNS packets, install a multicast listener to all available network interfaces.

Fixes #18748
commit 5049a049 (2023-11-17): net: mdns: Create a listener to all available network interfaces

Instead of just listening first network interface in the system,
install a multicast listener to all available network interfaces.

Fixes #18748