zephyrproject-rtos/zephyr #18749: Avenger96 regressed in mainline

Mani-Sadhasivam, 2019-08-28:
**Describe the bug**
I can't get the Avenger96 binary to work on the board anymore. The basic hello_world example itself is not working.

**To Reproduce**
Build and flash the binary to the board

**Additional context**
There were some ST specific clock changes done which affected STM32MP1, could that be a reason?

erwango, 2019-08-29:
@arnop2 , @yboutreux

yboutreux, 2019-08-29:
Hi @Mani-Sadhasivam,
hello_world is still working on stm32mp157c_dk2 with Zephyr latest commits.
MP1's clock_control driver itself doesn't seem to had a modification that could break it https://github.com/zephyrproject-rtos/zephyr/commits/master/drivers/clock_control/clock_stm32_ll_mp1.c
Do you have more informations about the specific clock changes ?

arnop2, 2019-08-29:
Hi Mani,
i tested the hello world sample on the avenger96 board: no issue on my side:

```
 root@stm32mp1-av96:~# cat /sys/kernel/debug/remoteproc/remoteproc0/trace0 
 ***** Booting Zephyr OS build v2.0.0-rc1-288-gcfd1f4950f7e *****
 Hello World! 96b_avenger96

```
My settings:
Cortex-A: Linux 5.3 + enable of the rproc in kconfig and dts
Cortex-M: Zephyr 2.0 rc1

Be carfull, if you use the kernel 5.3, you will now use the rproc driver that has been upstreamed. dts declaration must be adapted accordingly.

Mani-Sadhasivam, 2019-08-29:
@arnop2 @yboutreux 

Hmm. So it works with remoteproc driver in linux kernel but it is not working with u-boot as mentioned here: https://wiki.st.com/stm32mpu/wiki/How_to_start_the_coprocessor_from_the_bootloader

Earlier, I used to start the firmware using u-boot and it worked well. But with new binary it is not.

arnop2, 2019-08-29:
@Mani-Sadhasivam 

**when you say "new binary" is it the Uboot or the Zephyr binary?** 

If it works fine when you load zephyr elf file from Linux but ko from Uboot,  the zephyr binary is problably not the root cause. 
**Could you confirm that your Linux version is 5.3 ?**
In this case your issue is probably a misalignment betwen the DTS and yourUboot version.

you probably need to update the Uboot...  we are upstreaming rproc in Uboot based on the new dts description: https://patchwork.ozlabs.org/project/uboot/list/?series=122979&state. 
This version is compatible with the new way to describe rproc node in DTS.

Mani-Sadhasivam, 2019-08-29:
@arnop2 

> when you say "new binary" is it the Uboot or the Zephyr binary?

Zephyr binary.

> If it works fine when you load zephyr elf file from Linux but ko from Uboot, the zephyr binary is problably not the root cause.

I just tested the `v1.14.0` branch of Zephyr and it is working fine. So definitely something should've changed in Zephyr in between for STM32MP1. 

> Could you confirm that your Linux version is 5.3 ?

No, I'm using the vendor kernel (4.19.9)

arnop2, 2019-08-29:
@Mani-Sadhasivam 
Thanks for the clarification!
 I think i'm reproducing the issue on dk2 board. the firmware is runnuing without issue if it isloaded by the Linux, but trace is corrupted if it is loaded by the uboot loader. 
Main difference between both is the elf parser that is different between the Linux and the Uboot. Perhaps the delta between v1.14 and v2.0 could be explained by a linker update...? 
Need more in-depth investigation to understand the problem

arnop2, 2019-09-02:
@Mani-Sadhasivam 

We found an issue in Uboot. It is a side effect of adding the EXIDX program Header in the Zephyr elf file. 
The Uboot has a bug in the elf program header parser. In some cases the pointer to the headers is not incremented.
To fix this you need to recompile the Uboot after applying following patch:

@@ -430,11 +430,11 @@ static int rproc_load_elf_image(struct udevice *dev, unsigned long addr,
 		dev_dbg(dev, "Driver has no ops?\n");
 		return -EINVAL;
 	}
 
 	/* Load each program header */
-	for (i = 0; i < ehdr->e_phnum; ++i) {
+	for (i = 0; i < ehdr->e_phnum; ++i, phdr++) {
 		void *dst = (void *)(uintptr_t)phdr->p_paddr;
 		void *src = (void *)addr + phdr->p_offset;
 
 		if (phdr->p_type != PT_LOAD)
 			continue;
@@ -451,11 +451,10 @@ static int rproc_load_elf_image(struct udevice *dev, unsigned long addr,
 			       phdr->p_memsz - phdr->p_filesz);
 		flush_cache(rounddown((unsigned long)dst, ARCH_DMA_MINALIGN),
 			    roundup((unsigned long)dst + phdr->p_filesz,
 				    ARCH_DMA_MINALIGN) -
 			    rounddown((unsigned long)dst, ARCH_DMA_MINALIGN));
-		++phdr;
 	}
We will upstream the fix ASAP

Could you confirm that this patch solves your issue?

Thanks,
Arnaud

arnop2, 2019-09-02:
Sorry bad syntax for the patch...
```
@@ -430,11 +430,11 @@ static int rproc_load_elf_image(struct udevice *dev, unsigned long addr,
 		dev_dbg(dev, "Driver has no ops?\n");
 		return -EINVAL;
 	}
 
 	/* Load each program header */'
-	for (i = 0; i < ehdr->e_phnum; ++i) {'
+	for (i = 0; i < ehdr->e_phnum; ++i, phdr++) {
 		void *dst = (void *)(uintptr_t)phdr->p_paddr;
 		void *src = (void *)addr + phdr->p_offset;
 
 		if (phdr->p_type != PT_LOAD)
 			continue;
@@ -451,11 +451,10 @@ static int rproc_load_elf_image(struct udevice *dev, unsigned long addr,
 			       phdr->p_memsz - phdr->p_filesz);
 		flush_cache(rounddown((unsigned long)dst, ARCH_DMA_MINALIGN),
 			    roundup((unsigned long)dst + phdr->p_filesz,
 				    ARCH_DMA_MINALIGN) -
 			    rounddown((unsigned long)dst, ARCH_DMA_MINALIGN));
-		++phdr;
 	}

 	*entry = ehdr->e_entry;
 
 	return 0;
```

Mani-Sadhasivam, 2019-09-02:
@arnop2 Thanks for the patch! I will be on vacation for this week. Will test this patch asap.

mgiaco, 2019-09-16:
@arnop2  I am new to this topic. But is this fix needed if i would like to run code on the M4 via uboot only. So there is no running linux needed?

arnop2, 2019-09-16:
@mgiaco 
Yes this fix is needed if you want to load Zephyr 2.0 in Uboot. if loaded by Uboot, the remotre proc 
driver in the ST Linux Kernel distribution is able to attach to it (i.e. retrieve the resource table to enable trace and RPMsg).

arnop2, 2019-09-16:
@Mani-Sadhasivam 
Did you find time to test it?

Mani-Sadhasivam, 2019-09-16:
@arnop2 Not yet. Unfortunately my vacation just started this week, will try to test it whenever I have time. Thanks

erwango, 2019-11-01:
@Mani-Sadhasivam, did you had time to test?

Mani-Sadhasivam, 2019-11-01:
@erwango Sorry for not updating this issue. The problem got solved using @arnop2's patch.

closing...

closed 2019-11-01 by Mani-Sadhasivam