zephyrproject-rtos/zephyr #18778: soc: arm: nordic_nrf: Defining DT_..._DEV_NAME

MarekPieta, 2019-08-29:
**Is your enhancement proposal related to a problem? Please describe.**
Currently labels like e.g. `DT_GPIO0_DEV_NAME` for nordic_nrf are defined in `dts_fixup.h` even if the device is unavailable (defined as other label - `DT_NORDIC_NRF_GPIO_GPIO_0_LABEL`, which remains undefined). It makes us unable to use #ifdefs based on `DT_..._DEV_NAME`.

**Describe the solution you'd like**
Maybe mentioned labels could be defined only if device is available. This would make us able to perform checks using `#ifdefs` in applications (with current implementation, we would have to use internal labels like `DT_NORDIC_NRF_GPIO_GPIO_0_LABEL` or Kconfig labels).

galak, 2019-08-29:
Can you provide a more detailed example of what you are wanting to accomplish.

MarekPieta, 2019-08-29:
For example, I want to define array containing names of available gpio devices. If there is no gpio device with given index, I simply want to place NULL in the array (I wanted to achieve this using ifdefs based on e.g. `DT_GPIO0_DEV_NAME`). Currently I have to use Kconfig options for ifdefs - e.g. `CONFIG_GPIO_NRF_P0`.

I think, that leaving `DT_GPIO0_DEV_NAME` undefined when device is unavailable would be better solution, than `#define` using label (`DT_NORDIC_NRF_GPIO_GPIO_0_LABEL`), which is undefined .

ulfalizer, 2019-08-29:
@MarekPieta 
Would it be possible to get rid of the `dts_fixup.h` entry and use the generated devicetree macros directly, or does it get awkward for some reason?

MarekPieta, 2019-09-02:
@ulfalizer, to be honest I am not sure, why it was made this way (I assumed that there is a reason).
Other simple solution for this could be adding `ifdefs` to `dts_fixup` headers (but its not elegant one).

ulfalizer, 2019-09-06:
@MarekPieta 
Don't know either. Maybe the thinking was that it'd standardize how the fixing up of names is done.

I like the fix-it-yourself approach at least, but I'm pretty stone age and minimal-fanciness in how I do stuff. I was thinking of trying to get rid of `dts_fixup.h` files later.

galak, 2021-01-22:
Closing this as we've implemented a completely new/different devicetree api

closed 2021-01-22 by galak