zephyrproject-rtos/zephyr #18803: LTS - support time

markusr, 2019-08-30:
I [read](https://docs.zephyrproject.org/latest/development_process/release_process.html#long-term-support-lts) that a LTS release is made every 2 year. It is however not mentioned how long a LTS release will be supported.

It would be good to add a table with EOL-dates like [Ubuntu](https://wiki.ubuntu.com/Releases) or [linux kernel](https://www.kernel.org/category/releases.html) to the documentation.

The question came up in a discussion if we should base a new firmware which we are currently developing on v1.14 or on v2.0

markusr, 2019-09-27:
@galak 
So if I understand you correctly, both v1.14 and v2.0 will be supported until the next stable release v2.1 (which is also a LTS release).

aescolar, 2019-09-27:
@markusr v2.1 is not planned to be a LTS
v2.0 is not supported in the same way as the 1.14 LTS is. The 1.14 LTS does get many fixes backported (just not new features) while the 2.0 will not.

nashif, 2019-11-08:
LTS support will be 2 years, LTS (1.14) support will end in April 2021.

closed 2019-11-08 by nashif