zephyrproject-rtos/zephyr #18804: Channel Selection Algorithm Modification In Zephyr

BOZHENG001, 2019-08-30:
Hi All

I am now working in Channel Selection Algorithm (CSA), Bluetooth Low Energy on nRF52840DK. Basically, I would like to test both CSA 1 & 2, and also I would like to modify the CSA inside BLE, or even write my own CSA into BLE stack to test its performance.

I would like to do 3 things on nRF52840DK:

1. To switch between CSA 1 & 2
2. To control the channel map by software
3. To use modified / my own CSA to do some tests

I have already done the first 2 things on nRF52840DK by zephyr, and I also found some files inside zephyr source files about CSA, like "lll_chan.c". But it seems these files are not influencing channel selection process, because when I modified them, nothing happened to the process.

Could anyone please tell me if it is possible to modify CSA in zephyr?

carlescufi, 2019-08-30:
>like "lll_chan.c"

@BOZHENG001 

There are two Link Layers in Zephyr, the old, legacy one and the new split link layer. If you change code in `lll_*` files, you are modifying the split link layer. Make sure that when you build zephyr you enable it in Kconfig with `CONFIG_BT_LL_SW_SPLIT=y`

BOZHENG001, 2019-09-02:
> > like "lll_chan.c"
> 
> @BOZHENG001
> 
> There are two Link Layers in Zephyr, the old, legacy one and the new split link layer. If you change code in `lll_*` files, you are modifying the split link layer. Make sure that when you build zephyr you enable it in Kconfig with `CONFIG_BT_LL_SW_SPLIT=y`

closed 2019-09-02 by BOZHENG001

BOZHENG001, 2019-09-02:
@carlescufi 
Hi Carlescufi
Thanks for your information. Actually, I also tried that, enabling split link layer in Kconfig `(CONFIG_BT_LL_SW_SPLIT=y)`. Then I tried to modify "lll_chan.c" for both central and peripheral devices, but it did not work as i wished.
So i am confused if i need to configure more in Kconfig, or maybe i will have to rebuild the project with lll_* files myself?

carlescufi, 2019-09-02:
> @carlescufi
> Hi Carlescufi
> Thanks for your information. Actually, I also tried that, enabling split link layer in Kconfig `(CONFIG_BT_LL_SW_SPLIT=y)`. Then I tried to modify "lll_chan.c" for both [...]

Well the Link Layer and BLE doesn't support what you want to do natively. First I would suggest that you read the part of the spec that deals with CSA to understand how the current code is implemented. Once that is done you can start modifying it and we can probably help out a bit.

BOZHENG001, 2019-09-04:
@carlescufi 
Hi Carlescufi
Thanks for your suggestion! I am able to use split link layer now, but meeting with some new problems.
Basically, i enable the split link layer on both peripheral and central device, by setting: 

`CONFIG_BT_RECV_IS_RX_THREAD=y,
CONFIG_ENTROPY_GENERATOR=y,
CONFIG_BT_LL_SW_SPLIT=y`

But then, after CONNECT_REQ, my sniffer cannot update any more information between them. Also i tried to use my phone to connect with peripheral device, but they do not connect.
Could you please provide some advice?