zephyrproject-rtos/zephyr #18807: Support the Ubuntu Cross Toolchain

cfriedt, 2019-08-30:
**Is your feature request related to a problem? Please describe.**
It would be helpful for many Ubuntu users, and also for continuous integration (CI), to be able to use Ubunut's 'gcc-arm-none-eabi' cross-compiler for building zephyr for arm microcontrollers.

In particular, many CI loops use Ubuntu's cross-compiler because it is so easy to install from Ubuntu's default package repository in many containerized and non-containerized build environments.

It's as easy as 'apt-get update && apt install -y gcc-arm-none-eabi'.

Currently, I believe that only 'gnuarmemb' and 'crosstool-ng' toolchains are supported. In both cases, it's possible that downloading & installing them might require significantly more time and bandwidth than installing the ubuntu toolchain. In particular, Travis CI host their own Ubuntu package repository, and so it would be very quick to install from that source.

**Describe the solution you'd like**
Support for Ubuntu's toolchain along with a small bit of documentation on how to install it in the "3rd Party Toolchains" section.

**Describe alternatives you've considered**
I've used the gnuarmemb toolchain, but it does add some overhead to my CI process.

aescolar, 2019-08-31:
CC: @galak

nashif, 2019-09-05:
This is already supported in generic way using CROSS_COMPILE

https://docs.zephyrproject.org/latest/getting_started/toolchain_other_x_compilers.html?highlight=cross_compile

cfriedt, 2019-09-19:
Thanks @nashif - I can't believe I didn't see that beforehand!

closed 2019-09-19 by cfriedt