zephyrproject-rtos/zephyr #18813: fs: nvs: Cannot delete entries

menglingao1997, 2019-08-31:
when I first call `nvs_init()` function, _this time FLASH space ALL is 0xFF_.
then I write entry just like this:` nvs_write(&fs, 1, buf, 1);`
then I delete this entry: `nvs_delete(&fs, 1);`
then, I read:` nvs_read(&fs, 1, buf1, 1);` **Successful return**

```C
while (1) {
		rd_addr = wlk_addr;
		rc = nvs_prev_ate(fs, &wlk_addr, &wlk_ate);
		if (rc) {
			return rc;
		}
		if ((wlk_ate.id == id) && (!nvs_ate_crc8_check(&wlk_ate))) {
			break;
		}
		if (wlk_addr == fs->ate_wra) {
			break;
		}
	}

	if (wlk_addr != fs->ate_wra) {
```

just like above code show, when i call `nvs_delete`, the loop find prev entry,and finally find it, then break the loop, but `if (wlk_addr != fs->ate_wra) ` is true.

Laczen, 2019-08-31:
@menglingao1997, thank you for reporting this bug. I think it is indeed a bug. It only appears when you are trying to delete the just added entry (then wlk_addr == fs->ate_wra and wlk_ate.id == id).
A possible fix is to add a bool "prev_found" and set it to true when a valid id match is found:
line 785: `break;` replace by `prev_found = true; break;`
line 792: replaced by: `if (prev_found) {`

carlescufi, 2019-08-31:
> @menglingao1997, thank you for reporting this bug. I think it is indeed a bug. It only appears when you are trying to delete the just added entry (then wlk_addr == fs->ate_wra and wlk_ate.id == id). [...]

Thank you @Laczen for the suggested fix. I have posted a [PR](#18820) with your suggestion.

PR #18820 (2019-09-02): nvs: Fix deletion of the last entry added
Make sure that the last entry added is deleted correctly by storing the
fact that one was found in a local variable.

Fix by Laczen JMS <laczenjms@gmail.com>

Fixes #18813.
commit 6f5dc33e (2019-09-02): nvs: Fix deletion of the last entry added

Make sure that the last entry added is deleted correctly by storing the
fact that one was found in a local variable.

Fix by Laczen JMS <laczenjms@gmail.com>

Fixes #18813.
commit 3ad9394a (2019-09-25): nvs: Fix deletion of the last entry added

Make sure that the last entry added is deleted correctly by storing the
fact that one was found in a local variable.

Fix by Laczen JMS <laczenjms@gmail.com>

Fixes #18813.
closed 2019-09-02 by carlescufi