zephyrproject-rtos/zephyr #18821: Documentation: parent vs child in DeviceTree nodes

karstenkoenig, 2019-09-01:
In my understanding parent is used in node descriptions for bus devices. 
[spi-device.yaml](https://github.com/zephyrproject-rtos/zephyr/blob/master/dts/bindings/spi/spi-device.yaml) sets `parent: bus: spi` and is included by all sort of spi devices, such as sensors.

Where as [spi.yaml](https://github.com/zephyrproject-rtos/zephyr/blob/master/dts/bindings/spi/spi.yaml) sets `child: bus: spi`and is included by all actual SPI controllers/peripherals.

In the [binding template yaml](https://github.com/zephyrproject-rtos/zephyr/blob/0d766c668de052f00c04820897ca966cfd781aa4/dts/bindings/binding-template.yaml#L18) the explanation of parent and child seems to be the opposite of what it is used by implementations

Now do I just misunderstand the description? Or is the use of child and parent inverted in the actual yaml files using it?

karstenkoenig, 2019-09-01:
Ping @ulfalizer 
You did lots of DTS cleanup, maybe you can shed some light here :-)

ulfalizer, 2019-09-01:
Oops... I accidentally swapped them when adding the documentation.

Fix here: https://github.com/zephyrproject-rtos/zephyr/pull/18822

PR #18822 (2019-09-01): dts: binding-template.yaml: Fix 'parent/child: bus:' description
'child: bus:' should be in the binding for the bus node, and
'parent: bus:' in the binding for devices that appear on the bus.

The description accidentally swapped them. Fix it.

Fixes: #18821
commit e5974f72 (2019-09-01): dts: binding-template.yaml: Fix 'parent/child: bus:' description

'child: bus:' should be in the binding for the bus node, and
'parent: bus:' in the binding for devices that appear on the bus.

The description accidentally swapped them. Fix it.

Fixes: #18821
closed 2019-09-01 by galak