zephyrproject-rtos/zephyr #18852: west flash fails for cc1352r_launchxl

cfriedt, 2019-09-02:
**Describe the bug**

`west flash` fails in openocd for board `cc1352r1_launchxl`.

**To Reproduce**
Steps to reproduce the behavior:
1. `west build --pristine -b cc1352r1_launchxl samples/hello_world`
2. `west flash`
4. See error

**Expected behavior**
flashing should work

**Impact**
annoyance

**Screenshots or console output**

```
$ west flash
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner openocd
-- runners.openocd: Flashing file: /home/cfriedt/workspace/zephyrproject/zephyr/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.10.0+dev-00916-g42cee465-dirty (2019-06-19-19:21)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter_nsrst_delay: 100

Info : XDS110: connected
Info : XDS110: firmware version = 2.3.0.18
Info : XDS110: hardware version = 0x0023
Info : XDS110: connected to target via JTAG
Info : XDS110: TCK set to 2500 kHz
Info : clock speed 2500 kHz
Info : JTAG tap: cc13x2.jrc tap/device found: 0x3bb4102f (mfg: 0x017 (Texas Instruments), part: 0xbb41, ver: 0x3)
Info : JTAG tap: cc13x2.cpu enabled
Info : cc13x2.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* cc13x2.cpu         cortex_m   little cc13x2.cpu         halted

Info : JTAG tap: cc13x2.jrc tap/device found: 0x3bb4102f (mfg: 0x017 (Texas Instruments), part: 0xbb41, ver: 0x3)
Info : JTAG tap: cc13x2.cpu enabled
Error: Debug regions are unpowered, an unexpected reset might have happened
Error: JTAG-DP STICKY ERROR
Error: Could not find MEM-AP to control the core
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: DP initialisation failed

Info : XDS110: disconnected
ERROR: command exited with status 1: /usr/local/bin/openocd -f /home/cfriedt/workspace/zephyrproject/zephyr/boards/arm/cc1352r1_launchxl/support/openocd.cfg -c init -c targets -c 'reset halt' -c 'flash write_image erase /home/cfriedt/workspace/zephyrproject/zephyr/build/zephyr/zephyr.hex' -c 'reset halt' -c 'verify_image /home/cfriedt/workspace/zephyrproject/zephyr/build/zephyr/zephyr.hex' -c 'reset run' -c shutdown
```
**Environment (please complete the following information):**
 - OS: Ubuntu Bionic
 - Toolchain: GNU ARM Embedded 2019 q3
 - OpenOCD: commit 6ea43726a801baa718fd08dcdb8ae5835b8a2385
 - Commit (Zephyr): 0906671a7b17d7218c7f3acf0bcd991712dafd35

**Additional context**

The following command works perfectly.
```
openocd -f board/ti_cc13x2_launchpad.cfg -c "reset_config trst_only" -c "init" -c "reset halt" -c "flash write_image erase $PWD/build/zephyr/zephyr.elf"
```
It looks like all that that is missing is `-c "reset_config trst_only"`

cfriedt, 2019-09-02:
Yep. It's a one-liner. I'll create a PR.

mbolivar, 2020-02-11:
Discussion in #18854 indicates that this is no longer an issue, so closing. Feel free to reopen if I read that wrong.

closed 2020-02-11 by mbolivar