zephyrproject-rtos/zephyr #18858: Runner support for stm32flash utility

boretom, 2019-09-03:
Hi,

I got a 96b_wistrio board and it uses a STM32L1 MCU. Flashing the resulting firmware is done using the stm32flash open source utility.
To my surprise there is no runner for what I thought is a popular flashing tool. Searching the issues I didn't find any request for it.

**Current situation**
one has to manually use the stm32flash utility

**What I'd like**
Be able to use `west flash` to flash the firmware

**What I have done so far**
I implemented a basic runner for stm32flash - not tests yet - which runs fine for me but does only support a small subset of feature of the stm32flash utility (stm32flash-runner branch in my fork of your repo).

**My question**
With your knowledge of the Zephyr project, do you think there be an interested to have this runner extended to support all the feature with the idea that it may get included in upstream at some point?

I may add that I'm very new to embedded programming and therefore don't have much experience.
After reading a bit I realise that most people probably use ST-Link or BlackMagic Probe or something similar to flash and debug. That's maybe the reason no one really is using stm32flash.

carlescufi, 2019-09-03:
> With your knowledge of the Zephyr project, do you think there be an interested to have this runner extended to support all the feature with the idea that it may get included in upstream at some [...]

With nRF chips we support both raw J-Link and also `nrfjprog`, the Nordic flashing tool. I don't see why we wouldn't add support for `stm32flash`. A Pull Request is most welcome @boretom.

boretom, 2019-09-03:
Hey @carlescufi ,

Ok, I'll extend the runner and create a pull request when I'm ready.
Shall this ticket be closed?

mbolivar, 2019-09-03:
Thanks for the feature request! Adding @erwango since it affects STM32.

> Ok, I'll extend the runner and create a pull request when I'm ready.
> Shall this ticket be closed?

Please keep it open and make sure your pull request includes a Fixes line for it:

https://github.blog/2013-05-14-closing-issues-via-pull-requests/

boretom, 2019-09-05:
Hi @carlescufi,

Would be able to tell me or provide a link to how to run runner test? I can't seem to figure out how to invoke them.

/Thomas

carlescufi, 2019-09-05:
> Hi @carlescufi,
> 
> Would be able to tell me or provide a link to how to run runner test? I can't seem to figure out how to invoke them.
> 
> /Thomas

@boretom 
https://github.com/zephyrproject-rtos/zephyr/blob/master/scripts/west_commands/README.txt

boretom, 2019-09-05:
Thanks @carlescufi, not sure how I overooked it.

PR #18981 (2019-11-05): west: runner: Add runner stm32flash
Hi,

Here the PR regarding stm32flash runner (closes #18858). I of course agree to the Apache 2.0 license and the git commit are therefore signed off by me.

Calling this runner works by passing the `--action` parameter to define what should be done. Not passing any action defaults to `--action write`.

The `action` choices would be either `write` the flash binary, `start` the program execution, `info` to get the device information, to `erase` the flash region in size of the firmware binary.

The defaults are set that no parameters are needed. Naturally some [... cut]
closed 2019-11-05 by nashif