zephyrproject-rtos/zephyr #18865: Fatal Usage Fault When Bluetooth And OpenThread Are Enabled On NRF52840 Multiprotocol Support

thomasthep, 2019-09-03:
**Description**
Compile application with Bluetooth and OpenThread enabled causes a fatal error to occur.
This can be reproduced by using the example in samples/net/sockets/echo_server provided within Zephyr's repository. Use the overlay overlay_ot.conf and set CONFIG_BT=y.

Nordic NRF5 SDK supports multi-protocol and the open sourced code also does.
https://devzone.nordicsemi.com/nordic/nordic-blog/b/blog/posts/unique-threadbluetooth-multiprotocol-solution-with
https://github.com/NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver/wiki/Multiprotocol-support

**To Reproduce**
Steps to reproduce the behavior:
1. Copy sample
2. Enable Bluetooth in overlay_ot.conf
3. cmake -B build -GNinja -DBOARD=nrf52840 -DCONF_FILE=prj.conf -DOVERLAY_CONFIG=overlay-ot.conf 
4. ninja -C build flash
5. Serial UART and find console output (see below)

Make sure both CONFIG_BT=y And CONFIG_NET_L2_OPENTHREAD=y are set.

**Expected behavior**
Debug messages of protocols initialized and then shell. Something along the lines (example missing bt messages as it fails to initialize).
```
***** Booting Zephyr OS build v2.0.0-rc2 *****
[00:00:00.000,000] <inf> ieee802154_nrf5: nRF5 802154 radio initialized
[00:00:00.005,187] <inf> net_l2_openthread: State changed! Flags: 0x047f1319 Current role: 0
[00:00:00.005,645] <inf> net_l2_openthread: OpenThread version: OPENTHREAD/g0dbecf50; NONE; Sep  3 2019 11:22:50
[00:00:00.005,645] <inf> net_l2_openthread: Network name: ot_zephyr
[00:00:00.006,622] <inf> net_l2_openthread: State changed! Flags: 0x00000004 Current role: 1
[00:00:00.006,683] <inf> net_l2_openthread: State changed! Flags: 0x00000001 Current role: 1
[00:00:00.006,805] <inf> net_l2_openthread: State changed! Flags: 0x00001000 Current role: 1
[00:00:00.006,896] <inf> net_l2_openthread: State changed! Flags: 0x00001000 Current role: 1
[00:00:00.007,049] <inf> net_l2_openthread: State changed! Flags: 0x00000010 Current role: 1
[00:00:00.007,080] <inf> net_l2_openthread: State changed! Flags: 0x00000020 Current role: 1
[00:00:00.013,305] <inf> net_config: Initializing network
[00:00:00.013,458] <inf> net_l2_openthread: State changed! Flags: 0x00000001 Current role: 1
[00:00:02.009,246] <inf> net_l2_openthread: State changed! Flags: 0x00000040 Current role: 1
[00:00:03.013,641] <inf> net_echo_server_sample: App initialization

[00:00:03.013,641] <inf> net_echo_server_sample: Run echo server
[00:00:03.013,732] <inf> net_echo_server_sample: Waiting for UDP packets on port 4242 (IPv6)...
[00:00:04.453,063] <inf> net_l2_openthread: State changed! Flags: 0x000012a5 Current role: 4
uart:~$ <tab>
  bt       clear    flash    help     history  log      net      resize
  ot       shell    ticker
```

**Impact**
Cannot get Bluetooth to be used for provisioning and possibly debugging.
Bluetooth Mesh was also going to be added as fallback in the event the Thread protocol fails for any reason. Eventually a bridge might be required in the future to support multiple product lines and guarantee compatibility.

**Screenshots or console output**
```
***** Booting Zephyr OS build v2.0.0-rc2 *****
FATAL: ***** USAGE FAULT *****
FATAL:   Illegal use of the EPSR
FATAL: r0/a1:  0x00000000  r1/a2:  0x2000f490  r2/a3:  0x20007134
FATAL: r3/a4:  0x00000000 r12/ip:  0x00000007 r14/lr:  0x00021ff9
FATAL:  xpsr:  0x60000011
FATAL: Faulting instruction address (r15/pc): 0x00000000
FATAL: >>> ZEPHYR FATAL ERROR 0: CPU exception
FATAL: Current thread: 0x20007ec0 (unknown)
[00:00:00.000,000] <inf> ieee802154_nrf5: nRF5 802154 radio initialized
[00:00:00.014,831] <inf> net_l2_openthread: State changed! Flags: 0x047f1319 Current role: 0
[00:00:00.015,319] <inf> net_l2_openthread: OpenThread version: OPENTHREAD/g0dbecf50; NONE; Sep  3 2019 12:03:58
[00:00:00.015,350] <inf> net_l2_openthread: Network name: ot_zephyr
[00:00:00.016,387] <inf> net_l2_openthread: State changed! Flags: 0x00000004 Current role: 1
[00:00:00.016,448] <inf> net_l2_openthread: State changed! Flags: 0x00000001 Current role: 1
[00:00:00.016,571] <inf> net_l2_openthread: State changed! Flags: 0x00001000 Current role: 1
[00:00:00.016,693] <inf> net_l2_openthread: State changed! Flags: 0x00001000 Current role: 1
[00:00:00.016,815] <inf> net_l2_openthread: State changed! Flags: 0x00000010 Current role: 1
[00:00:00.016,845] <inf> net_l2_openthread: State changed! Flags: 0x00000020 Current role: 1
FATAL: Halting system
```

**Build Environment:**
 - OS: MacOS
 - Toolchain: Zephyr SDK
 - Commit: 2.0.0-rc2
 - Target: NRF52840

carlescufi, 2019-09-03:
@thomasthep We do **not** support multiprotocol in Zephyr. @rlubos can you please confirm?

rlubos, 2019-09-03:
@carlescufi @thomasthep Correct, multiprotocol solution available in NRF5 SDK is not supported in Zephyr.

thomasthep, 2019-09-03:
@carlescufi If this is so, I believe it should be mentioned within the documentation. I checked multiple times but did not find any such mentions.

@rlubos I just noticed this is rather specific to Nordic's SoftDevice. However, there is documentation on how the scheduling works and there is a Simulator Arbiter (https://github.com/NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver/tree/master/src/rsch/raal/simulator). I believe this could be more thoroughly investigated and have the radio scheduler (Radio Timeslot API) implemented in the future to support multiprotocol on NRF52840 with Zephyr.

carlescufi, 2019-09-03:
@thomasthep I will close this since this is not currently supported. Feel free to open a feature request GitHub issue. Thanks!

closed 2019-09-03 by carlescufi

thomasthep, 2019-09-04:
@carlescufi Sure thing. Thanks.