zephyrproject-rtos/zephyr #18873: zsock_socket() should support proto==0

kFYatek, 2019-09-03:
**Is your enhancement proposal related to a problem? Please describe.**
Most implementations of BSD-style sockets, both those aiming at comprehensive POSIX compliance (e.g. Linux, macOS, *BSD) and those intentionally deviating from it (e.g. WinSock, lwIP), allow calling `socket()` with the `proto` argument set to 0, which causes the protocol to be defaulted to a typical choice.

E.g. `socket(AF_INET, SOCK_STREAM, 0)` is equivalent to `socket(AF_INET, SOCK_STREAM, IPPROTO_TCP)` and `socket(AF_INET, SOCK_DGRAM, 0)` is equivalent to `socket(AF_INET, SOCK_DGRAM, IPPROTO_UDP)`.

This is actually mandated by POSIX (see [here](http://pubs.opengroup.org/onlinepubs/9699919799/functions/socket.html) and [here](http://pubs.opengroup.org/onlinepubs/9699919799/functions/V2_chap02.html#tag_15_10_18)), but as I mentioned above, even implementations that do not aim for full POSIX compliance typically support this usage.

This makes it harder to port some libraries and applications that would otherwise be trivially portable to Zephyr, but use `socket()` in this way.

**Describe the solution you'd like**
I'd like `zsock_socket()` to behave as POSIX-compliant `socket()` would for this case. A minimal viable solution would be to make (AF_INET|AF_INET6, SOCK_STREAM) default the protocol to `IPPROTO_TCP` and (AF_INET|AF_INET6, SOCK_DGRAM) default the protocol to `IPPROTO_UDP`, as these are the two most common cases.

jukkar, 2019-09-03:
Sure, this makes sense. I will prepare a PR for this.

pfalcon, 2019-09-05:
This is actually a bug, marking as such.

commit d72212ad (2019-09-25): net: socket: Set default protocol if proto == 0

If proto field in socket() call is set to 0, then we should have
a sane default for it that depends on the type of the socket.

Fixes #18873