zephyrproject-rtos/zephyr #18896: Concurrent Multi-Protocol Support NRF52840

thomasthep, 2019-09-04:
Zephyr should support multi-protocol handling on NRF52840 chipsets. This would allow the use of Bluetooth LE/5 and IEEE802.15.4 PHY radios simultaneously, possibly also others such as 2.4GHz proprietary and ANT.

This would allow communication with multiple devices enabling easier configuration and possibly also provide a backup network.

Scenarios include but are not limited to:
* Provisioning OpenThread to a network using a handheld device such as a smartphone which is Bluetooth LE/5 enabled.
* Provide a backup/fallback communication layer for OpenThread switching to e.g. Bluetooth Mesh
* Act as (mini) bridge for non OpenThread devices
* Direct device control from a smartphone
* Bluetooth LE/5 scanning and delivering data via OpenThread (e.g. asset tracking, indoor positioning)

NordicSemiconductors' SoftDevice within the SDK 3.1.0 supports concurrent multiprotocol.
https://www.nordicsemi.com/Software-and-Tools/Software/nRF5-SDK-for-Thread-and-Zigbee

There's more documentation on how this works.
https://github.com/NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver/wiki/Multiprotocol-support
https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.s140.api.v5.0.0%2Fgroup___n_r_f___s_o_c___f_u_n_c_t_i_o_n_s.html&anchor=ga55a52eb4d0491cb7105de6a13eb2b11b
https://github.com/NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver/tree/master/src/rsch

After some digging, I found that the Zephyr Bluetooth driver for Nrf uses the radio directly via the hal abstractions.
https://github.com/zephyrproject-rtos/zephyr/blob/ec2f50363500ab9938c594cacf85f7db1c07d232/subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/radio/radio_nrf52840.h

Nordic's SoftDevice includes a Radio Event Manager (REM) which provides a Timeslot API. This is then used within the SoftDevice implementation of the Radio Arbiter Abstraction Layer (RAAL) which allows the software to request radio time slots.

Concluding from the given information, I guess the bluetooth driver will have to be abstracted, a REM implemented (possibly with a custom Timeslot API) to allow concurrent multi-protocol to be used within Zephyr on the NRF52840.

Any additions, clarifications, alternative solutions and possibly even pull requests are welcome.

Initially filed a bug ticket https://github.com/zephyrproject-rtos/zephyr/issues/18865

aescolar, 2019-09-04:
CC @carlescufi

bfrog, 2019-09-16:
Being able to use thread and ble or ant and ble at the same time would be really interesting for me as well for some future product ideas I have

thomasthep, 2019-09-25:
@bfrog I'm working on a small proof of concept. As soon as I get anywhere near some usability, I'll create a branch.
@carlescufi Would it make sense to also create documentation for the SoCs similar to 'supported boards'? The SoC is basically the main building block and boards would have peripherals and other features on top, or some other features might be disabled (such as pins not routed on the PCB). That's something I have been missing when going through the docs.

I've been busy with work lately and work on this when I get some free time on my hands.

carlescufi, 2020-03-24:
@thomasthep on the Nordic side there are no plans to support this in the near future.

thomasthep, 2020-03-24:
A shame but guess there aren't so many use cases out there requiring both anyways. If there are, you could always add another SoC.

I myself have been messing around with implementing a scheduler as described in the SDK. It would require a whole lot work to change the driver layer and possibly major changes in the architecture.

The hacked up version of mine works to some degree, but timing as well as race conditions are a pain.

Let's see what the future beholds.

farhangj, 2021-01-04:
Any movement on this ticket? it would be really helpful for my use-case to have Thread+BLE and a FEM support.

carlescufi, 2021-02-12:
Nordic has [multiprotocol](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/ug_multiprotocol_support.html) and [FEM](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/ug_radio_fem.html) support in [nRF Connect SDK](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/index.html), which is based on Zephyr. Take a look at that.

farhangj, 2021-02-12:
@carlescufi which version? 1.4.2 or 1.5.0? I don't see it in the release notes or any commits specifying FEM with Multiprotocol support. I'd be interested to see the actual commit for this feature. Thanks!

carlescufi, 2021-02-15:
> @carlescufi which version? 1.4.2 or 1.5.0? I don't see it in the release notes or any commits specifying FEM with Multiprotocol support. I'd be interested to see the actual commit for this feature. Thanks!

Sorry, I should've been more specific. It's in 1.5.0, which will be released in the next few days.

markus-becker-tridonic-com, 2021-07-26:
@carlescufi i believe this issue can be closed now.

closed 2021-08-17 by carlescufi