zephyrproject-rtos/zephyr #18906: Problem on build when calling objcopy to generate isrList.bin

kaiofking, 2019-09-04:
Bug description:
On build, the generation of isrList.bin return an error. 
It uses system objcopy (/usr/bin/objcopy).
After try with arm-none-eabi-objcopy, it works in this case. I think script uses the wrong objcopy.

To Reproduce:
ninja or make on a generated build folder.

Expected behavior:
isrList.bin is generated.

Impact: 
Build is broken.

Console output:
```
objcopy --input-target=elf32-littlearm --output-target=binary --only-section=.intList /home/robin-charles/firmware/applications/screen/build/zephyr/zephyr_prebuilt.elf isrList.bin
objcopy:/home/robin-charles/firmware/applications/screen/build/zephyr/zephyr_prebuilt.elf: invalid bfd target
```

Environment:
 - OS: Ubuntu 19.04
 - Toolchain: GCC ARM 8 (2019Q3)
 - 2.0.0rc2

aescolar, 2019-09-05:
CC @SebastianBoe

SebastianBoe, 2019-09-05:
Hi, please post the:

Board,
Sample,
git revision,
ZEPHYR_TOOLCHAIN_VARIANT

SebastianBoe, 2019-09-05:
Could not reproduce.

On my machine "arm-none-eabi-objcopy" is used.

kaiofking, 2019-09-05:
Board is nucleo_f413zh
ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb
No explicit ZEPHYR_SDK_INSTALL_DIR definition.
GNUARMEMB_TOOLCHAIN_PATH="$ENV{ZEPHYR_BASE}/../toolchain/gcc-arm-none-eabi-8-2019-q3-update"

After some tries, it seems CMAKE_OBJCOPY is already set before find_program(CMAKE_OBJCOPY ...) is invoked.
I don't know where it could be set before.

I remove all build cache files at all build folder generation.

SebastianBoe, 2019-09-05:
I believe that the wrong CMAKE_OBJCOPY version is coming from the previous CMakeCache.txt version.

Can it be reproduced with these commands for instance?

```
cd $ZEPHYR_BASE/samples/hello_world
grep CMAKE_OBJCOPY . -r && rm -rf oe && cmake -H. -Boe --trace-expand &> dump.txt && grep CMAKE_OBJCOPY . -r
```

kaiofking, 2019-09-05:
Ok it helps a bit. In dump.txt find_program(CMAKE_OBJCOPY ...) is called many times before "the good call" (aka zephyr/cmake/bintools/gnu/target.cmake one).

Before this call, it is called twice by (..)/cmake-3.14/Modules/CMakeFindBinUtils.cmake.

So it is due to my installation, but do you know why this cmake module is called before ?

SebastianBoe, 2019-09-05:
On my machine zephyr/cmake/bintools/gnu/target.cmake is the first call.

Please post the full output of

grep CMAKE_OBJCOPY dump.txt

or, if possible, upload dump.txt somewhere.

SebastianBoe, 2019-09-05:
I think CMakeFindBinUtils is indirectly invoked by 'project'.

Could you share the app CMakeLists.txt.

If project is invoked before boilerplate.cmake there is a problem.

kaiofking, 2019-09-05:
CMakeLists.txt:
`cmake_minimum_required(VERSION 3.8.2)

# Define project
project(screen)

# Set board settings
set(ENV{BOARD} "nucleo_f413zh")

# Modules inclusions
set(USE_MODULE_LOG ON)
set(USE_MODULE_HAL_STM32 ON)
set(USE_MODULE_COMMAND_ENGINE ON)
set(USE_MODULE_I2C_LAYER ON)

# Include common sources
include(../applications.cmake)

# Add includes
target_include_directories(app PRIVATE include/)

# Add sources
target_sources(app PRIVATE src/main.c)
target_sources(app PRIVATE src/command.c)
target_sources(app PRIVATE src/circle_led_manager.c)`

and then boilerplate.cmake is invoked in applications.cmake:
`# Set toolchainconfiguration
set(ENV{ZEPHYR_TOOLCHAIN_VARIANT} "gnuarmemb")
set(ENV{GNUARMEMB_TOOLCHAIN_PATH} "$ENV{ZEPHYR_BASE}/../toolchain/gcc-arm-none-eabi-8-2019-q3-update")

# Include boilerplate
include($ENV{ZEPHYR_BASE}/cmake/app/boilerplate.cmake NO_POLICY_SCOPE)

# Include modules
include($ENV{ZEPHYR_BASE}/../modules/modules.cmake)`

So you're right project is invoked before. I have tried and it works now.

Thanks a lot for your reactivity ! :)

SebastianBoe, 2019-09-05:
Great.

closed 2019-09-05 by SebastianBoe