zephyrproject-rtos/zephyr #18927: settings: deprecte base64 encoding

nvlsianpu, 2019-09-05:
Base64 encoding is something we inherited form Mynewt `config` subsystem. It was here for
allowing to keep many key-value pairs in one file.
As key-value pairs encoding was changes during many reworks we had, base64 encoding
is not required anymore for any backend we have (FS, FCB, NVS).

Deprecation of base64 encoding will simplify the code, and remove the need for maintaining something on which we have no gain today.

**preferred solution**
deprecation

**alternatives**
deprecation + extract base64 encoding as stream codec to independent utility module.

nvlsianpu, 2019-09-05:
Need your voice @Laczen @carlescufi @jhedberg @rakons @kapi-no

aescolar, 2019-09-05:
CC @fnde-ot

carlescufi, 2019-09-05:
I agree with the deprecation proposal

jhedberg, 2019-09-05:
Sounds good to me

rakons, 2019-09-05:
Base64 encoding is something that should be handled in a backend if it requires it. Moreover - if the backend uses base64 or stores data in binnary form should be transparent from the settings module perspective. I agree we should simplify the settings module code.

We do not need to extract base64 encoding as we have base64 library in the lib/os directory.

kapi-no, 2019-09-05:
I`m also in favour of it.

Any simplification of the settings module is welcome :)

Laczen, 2019-09-05:
If I remember correctly there are some users that have units in the field that depend on base64 encoding (with fcb). Although I am also in favour of the simplification  support for these users is needed.Making a special fcb backend with support for base64 encoding might be a solution. When simplifying please remove the shared code between nffs and fcb, this is what really complicates the module.

nvlsianpu, 2019-09-06:
@fnde-ot What is your opinion on the deprecation.

aescolar, 2019-09-06:
@fnde-ot has some temporary issues accessing his GitHub account.

But this is his answer (with which I concur):
"I of course agree with the proposed changes to deprecate base64 encoding from settings subsystem. This should be a backend thing."

Laczen, 2019-09-06:
Maybe some clarification is needed: the base64 encoding is used by the settings module to define how the variable-data pair is stored in the backend. It is only used to define how the data is stored, not how the pair is stored. It is not possible to move it to the "backend" (make it part of fcb, nffs, ...) while still keeping compatibility with existing users.
It is possible to move it from the shared code for fcb/nffs support (in the settings module) to separate code for fcb/nffs (in the settings module) while maintaining compatibility (e.g. only for fcb).

So I guess the real question is: do we decide to drop base64 encoding and leave users in a situation where they can no longer use their existing stored data ? Would it be possible to define some upgrade scheme for these users (e.g support base64 backend only for reading and let them write to a different area (without base64) for new storage) ?

nvlsianpu, 2019-09-06:
@DeclanTraill any thoughts?

rakons, 2019-09-06:
Backend gets struct  settings_store,  name, value and val size. Backend uses settings_line_write where the base64 conversion happens.  I do  not  see any reason why  cannot move it directly into fcb backend to keep the support for the users.

Laczen, 2019-09-06:
@rakons, I agree with your proposal. Moving the settings_line_write into the settings fcb backend and removing the shared code between fcb and nffs (this is mainly settings_line_write) would certainly simplify things a lot.

Depreciating base64 encoding is IMO something else, this is keeping the shared code between fcb and nffs and removing base64 encoding from both. This would have a minor effect on complexity and possibly effect some users that are using base64 encoding.

nwsetec, 2019-09-09:
I'm working on a product that began based on 1.13 (then base64 encoding was the standard for FCB) and now with a few workarounds for values that were stored as strings it is now based on 1.14 (and providing updates to 1.13 based devices in the field).  It is unlikely the product will be upgraded to beyond 1.14 but if we found the new split layer BLE controller to perform better then there's a small chance it might happen.  It would make it much harder for us to do this without base64 support for FCB or a way to update the encoding. Thanks @Laczen ;)

Having said that I can definitely understand the want/need to keep things simple.

We almost used this feature but I always thought that a settings file that could be sent over BLE SMP to nffs and loaded on boot and was partially human readable/verifiable in a text editor (that it's formed reasonably correctly) because binary data was base64 encoded was neat.  I know things have moved away from that.  I'm not sure how significant the RAM, flash and NV storage gains are from the changes but it does make things less accessible, at least for that particular usage.

de-nordic, 2019-12-06:
Here: https://github.com/de-nordic/zephyr/tree/settings-drop-base64 is my proposal how to deal with dropping the Base64 with settings, for now only with FS backends. I am still working on tests, so the test set is not complete, but the supporting code is.

The solution I would like to propose is that when Base64 deprecate configuration is on (`CONFIG_SETTINGS_DEPRECATE_USE_BASE64`), then only the Base64 decoding support will be left with settings, and every read will check for `_sys/nob64=1` marker line that marks the end of Base64 encoded lines.

When write occurs, first it will be checked if such marker already exists and if not,  it will be added and its position stored with settings_file structure. After this marker, every write is done without Base64 encoding.
When new file is created the marker will be the first item written.

Pros are:
1) Data is recoded to raw only on write;
2) Same API used; user does not need to know if data is raw on base64 encoded;
3) Single API reduces impact on code section;

Obvious cons of above solution are:
1) Additional check for marker on each read until marker is found; this impact gets reduced when marker becomes first line of settings file;
2) Additional line  (marker) added to each settings file;
3) Need for additional off_t variable within settings_file structure to store marker offset; need to initialize this variable when settings_file structure is defined;
4) Need for additional off_t variable within line_entry_ctx for temporal storage of marker offset

Laczen, 2019-12-06:
@de-nordic, I like the idea.
Since settings files are read linearly you could set the "base64encoding" on at every read/write start and switch off after you have found the marker, then there would be no requirement to add the location to the settings_file structure. You could simply add "base64enc_end" as a line to the file at first start.

de-nordic, 2019-12-06:
> @de-nordic, I like the idea.
> Since settings files are read linearly you could set the "base64encoding" on at every read/write start and switch off after you have found the marker, then there would [...]

@Laczen This was original idea and I might consider it as alternative. It worked fine for writes, since after one write you knew that there is marker already. The offset is stored for reads so that there is no strcmp done (in search of marker) if marker position is already known.

nvlsianpu, 2019-12-06:
@de-nordic Like the idea as wel. So for FCB back-end this should work as well (just record with the end of base64 section marker). FCB are read from the oldest to the newest item, right?

de-nordic, 2019-12-06:
@nvlsianpu it might be working too. I have not yet get myself familiar with FCB.

de-nordic, 2019-12-06:
@Laczen I think that I could do without offset. As I have said it is used to avoid calling strcmp in case if marker has already been found, but I can do another test that will limit strcmp calls, simply by checking if first letter of name read is _ before doing strcmp. It does not differ that much from comparing offset. The _ test would happen every time but so does the offset test.
Unfortunately it would not save us storage that much as bool might be size of off_t, specially if structure size is not packed, or bool is 32bit int and off_t is 32bit long.
There is also case with duplicate search, which is forward search that needs to ignore the marker.

Laczen, 2019-12-06:
> @Laczen I think that I could do without offset. As I have said it is used to avoid calling strcmp in case if marker has already been found, but I can do another test that will limit strcmp calls, [...]

This might not be a good idea: we should leave the possibility that data is stored with a name starting with _
> Unfortunately it would not save us storage that much as bool might be size of off_t, specially if structure size is not packed, or bool is 32bit int and off_t is 32bit long.

Is there a need to store this globally (in settings_file struct) ? Would you not pass the encode option to in line_entry_ctx ?

> There is also case with duplicate search, which is forward search that needs to ignore the marker.

Yes, the duplicate search does not need this. Do not try to over optimize, it is only to support a temporary situation where the marker is not at the start.

@nvlsianpu, this idea will not work for fcb. Compression will move the marker in the fcb filesystem.

de-nordic, 2019-12-06:
Sorry I managed to destroy @Laczen comment, I have restored it just now.

de-nordic, 2019-12-06:
>   This might not be a good idea: we should leave the possibility that data is stored with a name starting with _

I am still going to strcmp to marker, but not with strings that lack _ as first character.

>    BTW don't forget you only need to do strcmp if the bool has not been set.

That is exactly what I have been doing. If the offset has been set to non-OFFSET_MAX value, then I would strcmp in search for marker. If offset has been set to position of marker, then I would skip compare as the position of marker has been known.

 >   Is there a need to store this globally (in settings_file struct) ? Would you not pass the encode option to in line_entry_ctx ?

Yes I think that it might be possible as writing to file does load its contents in search for duplicate, which means that we may find marker just before writing data to file.

de-nordic, 2019-12-06:
@Laczen , @nvlsianpu we may repurpose padding in fcb structure as flags and flag FCB as converted to non-base64 during compression.

de-nordic, 2019-12-06:
> 
> 
> @Laczen , @nvlsianpu we may repurpose padding in fcb structure as flags and flag FCB as converted to non-base64 during compression.

Sorry my bad there is no padding byte... seen it somewhere else.

Laczen, 2019-12-06:
@de-nordic, I left some comment in your code (only for the read).

de-nordic, 2019-12-09:
> 
> 
> @de-nordic, I left some comment in your code (only for the read).

Thx

de-nordic, 2020-01-13:
If you are potential uses of the feature or want to have your input, please look at the review of a proposed solution here: https://github.com/zephyrproject-rtos/zephyr/pull/21657 , thanks.

de-nordic, 2020-06-22:
This has been fixed here: https://github.com/zephyrproject-rtos/zephyr/pull/21657
And here is complete removal PR: https://github.com/zephyrproject-rtos/zephyr/pull/26053/

closed 2020-11-06 by nvlsianpu