zephyrproject-rtos/zephyr #18934: Update Documentation To Reflect No Concurrent Multi-Protocol

thomasthep, 2019-09-05:
As a user, I would like to know that the project as of currently does not support concurrent multiprotocol as stated in https://github.com/zephyrproject-rtos/zephyr/issues/18865.

Update the documentation to state the status of this.

carlescufi, 2019-09-05:
@thomasthep would you mind sending a Pull Request?

thomasthep, 2019-09-05:
@carlescufi Will do.
Are there any other SoCs which support 802.15.4 that are currently supported by Zephyr?
If yes, I assume they would also use the same on-chip radio which would cause this same issue.

carlescufi, 2019-09-05:
> @carlescufi Will do.
> Are there any other SoCs which support 802.15.4 that are currently supported by Zephyr?

There is one from Nordic with the same radio (nRF52811) and then there's a bunch of SoCs with 15.4 support, but not all of them support other protocols. A quick list can be found here:

https://github.com/zephyrproject-rtos/zephyr/tree/master/drivers/ieee802154

> If yes, I assume they would also use the same on-chip radio which would cause this same issue.

As I mentioned, only the 52811.

thomasthep, 2019-09-25:
@carlescufi Would it make sense to also create documentation for the SoCs similar to 'supported boards'? The SoC is basically the main building block and boards would have peripherals and other features on top, or some other features might be disabled (such as pins not routed on the PCB). That's something I have been missing when going through the docs.

Adding it to the boards makes it redundant and there really isn't any good place to put this sort of info.

carlescufi, 2019-09-27:
> @carlescufi Would it make sense to also create documentation for the SoCs similar to 'supported boards'? The SoC is basically the main building block and boards would have peripherals and other [...]

Yes, but we don't really have a place for documenting the SoCs themselves today. We only have documentation for boards, not SoCs, hence my suggestion. Adding a new section about SoCs is certainly and option but would require more work and buy-in from the community. I am happy to review a PoC PR in that direction.

carlescufi, 2021-03-11:
Closing this in favor of #25663

closed 2021-03-11 by carlescufi