zephyrproject-rtos/zephyr #18935: [Zephyr 1.14] drivers: flash: spi_nor: Problematic write with page boundaries

gon1332, 2019-09-05:
**Describe the bug**
The driver is known to have an issue when the size of the chunk to be written is bigger than the page boundaries. It is fixed for the later versions of Zephyr through https://github.com/zephyrproject-rtos/zephyr/pull/17691 and https://github.com/zephyrproject-rtos/zephyr/pull/17994, but not for Zephyr v1.14.

In my opinion https://github.com/zephyrproject-rtos/zephyr/pull/17994 should be backported as a whole.

This issue has been created to backport the needed fixes in v1.14.

**To Reproduce**
Write an application that writes at unaligned page addresses.

**Expected behavior**
The driver should handle correctly writes across page boundaries.

**Impact**
Showstopper

**Screenshots or console output**

**Environment (please complete the following information):**
 - OS: GNU/Linux
 - Toolchain: Zephyr SDK
 - v1.14-branch

gon1332, 2019-09-05:
@pabigot Here you are.

pabigot, 2019-10-15:
Fixed in 1.14.1

PR #19229 (2019-09-25): [Backport v1.14-branch] drivers: spi_nor: fix writes across page boundaries
Closes #18935

### drivers/spi_nor: fix writes across page boundaries

The previous code limited the length of a write to the size of a page,
but did not check whether the starting position was far enough into the
page that the write would still cross a page boundary.

NOTE: SPI NOR page size is fixed as 256 regardless of what Kconfig says.

Backport: 2.0 660149b50073d4bfcead75153eb9082811df9391
Backport: 2.0 86dcf3c173e6a9542e7751fc2ab0850ad8d80943

### drivers/spi_nor: de-pessimize reads

SPI NOR devices require that [... cut]
closed 2019-10-15 by pabigot