zephyrproject-rtos/zephyr #18940: Counter External Trigger

bfrog, 2019-09-05:
**Is your feature request related to a problem? Please describe.**
I'd like to use the externally driven trigger for counters (pin edge causes counter to
increment) for frequency counting in a hardware agnostic way.

**Describe the solution you'd like**
I would've expected the counter API to provide the ability to allow for externally
driven counters as most hardware timers support this.

**Describe alternatives you've considered**
I can of course use the vender specific driver libraries as needed, and I will be for
an upcoming project. I'd like to do it in a vendor agnostic way of course.

carlescufi, 2019-09-05:
@nordic-krch could you please comment on this feature request and see how it aligns with the current counter API design?

henrikbrixandersen, 2020-01-11:
As far I see it, that is already fully supported by the current counter API. It is just a matter of having a driver, that configures the underlying counter hardware for counting external events (I have implemented a few these out-of-tree using the counter API already).

carlescufi, 2020-03-24:
Closing this based on @henrikbrixandersen 's assessment, please reopen if incorrect.

closed 2020-03-24 by carlescufi