zephyrproject-rtos/zephyr #18957: NET_L2: offload drivers aren't assigned a net_l2 which causes a crash in net_if_up()/net_if_down()

mike-scott, 2019-09-05:
**Describe the bug**
When using a network offloaded driver (CONFIG_NET_OFFLOAD / CONFIG_NET_SOCKETS_OFFLOAD) a net_l2 is never assigned to the net_if.
Several places reference the iface's net_l2 in the net_if code:
    l2_flags_get()
    net_if_up()
    net_if_down()

There are no checks here to make sure net_l2 isn't NULL and this results in an exception:
```
FATAL: ***** Reserved Exception ( -16) *****                                                                                   
FATAL: r0/a1:  0x00000010  r1/a2:  0x0000644f  r2/a3:  0x00000000                                                              
FATAL: r3/a4:  0x00000000 r12/ip:  0x2000474c r14/lr:  0x0001475b                                                              
FATAL:  xpsr:  0x00000000                                                                                                      
FATAL: Faulting instruction address (r15/pc): 0x0001b1cd                                                                       
FATAL: >>> ZEPHYR FATAL ERROR 0: CPU exception                                                                                 
FATAL: Current thread: 0x20004c4c (unknown)                                                                                    
```

**To Reproduce**
Steps to reproduce the behavior:
1. Build the LwM2M client w/ the SparkFun SARA-R4 enabled:
```
west build -b frdm_k64f -s zephyr/samples/net/lwm2m_client -- \
  -DSHIELD=sparkfun_sara_r4 \
  -DCONFIG_NET_CONFIG_PEER_IPV4_ADDR=\"198.199.108.159\" \
  -DCONFIG_NET_IPV6=n -DCONFIG_NET_CONFIG_NEED_IPV6=n \
  -DCONFIG_MODEM_UBLOX_SARA_R4_APN=\"hologram\" \
  -DCONFIG_MODEM_UBLOX_SARA_R4_MANUAL_MCCMNO=\"310410\" \
  -DCONFIG_COAP_INIT_ACK_TIMEOUT_MS=10000 -DCONFIG_MODEM_SHELL=y \
  -DCONFIG_LWM2M_SERVER_DEFAULT_PMIN=1 -DCONFIG_LWM2M_SERVER_DEFAULT_PMAX=360 \
  -DCONFIG_MODEM_LOG_LEVEL_DBG=y
```
2. `west flash`
3. See error after the modem intialization:
```
FATAL: ***** Reserved Exception ( -16) *****                                                                                   
FATAL: r0/a1:  0x00000010  r1/a2:  0x0000644f  r2/a3:  0x00000000                                                              
FATAL: r3/a4:  0x00000000 r12/ip:  0x2000474c r14/lr:  0x0001475b                                                              
FATAL:  xpsr:  0x00000000                                                                                                      
FATAL: Faulting instruction address (r15/pc): 0x0001b1cd                                                                       
FATAL: >>> ZEPHYR FATAL ERROR 0: CPU exception                                                                                 
FATAL: Current thread: 0x20004c4c (unknown)                                                                                    
```

**Expected behavior**
We shouldn't crash.

**Impact**
Oddly, this error doesn't always pop up like you would think it should.  Basically, it depends on the user config.  This is a show stopper for offloaded HW on certain SoCs at the moment.

**Environment (please complete the following information):**
 - OS: Arch Linux
 - Toolchain: Zephyr SDK 0.10.3
 - Commit SHA: v2.0.0-rc3

jukkar, 2019-09-06:
I think we can postpone this to 2.1

mike-scott, 2019-09-06:
@ioannisg @jukkar I'm ok with postpone.  It's fixing a crash but I had an odd time seeing it all the time.

ioannisg, 2019-09-06:
> 
> 
> @ioannisg @jukkar I'm ok with postpone. It's fixing a crash but I had an odd time seeing it all the time.

@mike-scott there will be a 2.0.1 milestone for fixes.

closed 2019-09-09 by nashif