zephyrproject-rtos/zephyr #18959: [Coverity CID :203907]Parse warnings in /include/bluetooth/conn.h

aasthagr, 2019-09-06:
Static code scan issues seen in File: /include/bluetooth/conn.h
 Category: Parse warnings
 Function: 
 Component: Bluetooth
 CID: 203907
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-09-06:
```
*** CID 203907:  Parse warnings  (PARSE_ERROR)
/include/bluetooth/conn.h: 292 in ()
286     	BT_SECURITY_L2,
287     	/** Level 3: Encryption and authentication (MITM). */
288     	BT_SECURITY_L3,
289     	/** Level 4: Authenticated Secure Connections and 128-bit key. */
290     	BT_SECURITY_L4,
291     
>>>     CID 203907:  Parse warnings  (PARSE_ERROR)
>>>     {CovLStr{v1}{expected a "}"}}
292     	BT_SECURITY_NONE   __deprecated = BT_SECURITY_L0,
293     	BT_SECURITY_LOW    __deprecated = BT_SECURITY_L1,
294     	BT_SECURITY_MEDIUM __deprecated = BT_SECURITY_L2,
295     	BT_SECURITY_HIGH   __deprecated = BT_SECURITY_L3,
296     	BT_SECURITY_FIPS   __deprecated = BT_SECURITY_L4,
```

jhedberg, 2019-09-06:
Marked as false positive in Coverity - it seems to get confused by the `__deprecated` annotation

closed 2019-09-06 by jhedberg