zephyrproject-rtos/zephyr #18960: [Coverity CID :203908]Error handling issues in /lib/libc/newlib/libc-hooks.c

aasthagr, 2019-09-06:
Static code scan issues seen in File: /lib/libc/newlib/libc-hooks.c
 Category: Error handling issues
 Function: _sbrk
 Component: libc
 CID: 203908
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-09-06:
```
*** CID 203908:  Error handling issues  (CHECKED_RETURN)
/lib/libc/newlib/libc-hooks.c: 247 in _sbrk()
241     static LIBC_DATA SYS_SEM_DEFINE(heap_sem, 1, 1);
242     
243     void *_sbrk(int count)
244     {
245     	void *ret, *ptr;
246     
>>>     CID 203908:  Error handling issues  (CHECKED_RETURN)
>>>     Calling "sys_sem_take" without checking return value (as is done elsewhere 9 out of 11 times).
247     	sys_sem_take(&heap_sem, K_FOREVER);
248     
249     #if CONFIG_NEWLIB_LIBC_ALIGNED_HEAP_SIZE
250     	ptr = heap_base + heap_sz;
251     #else
252     	ptr = ((char *)HEAP_BASE) + heap_sz;
```

aescolar, 2019-09-06:
Seems introduced in aed767a98c0

ioannisg, 2019-09-06:
Adding low here, since this is (just) the fact that we don't check the return value

PR #27132 (2020-07-25): Address multiple Coverity warning bugs
Multiple coverity issues are addressed in this PR.

Fixes #26985 
Fixes #26986 
Fixes #25724
Fixes #25726 
Fixes #25790 
Fixes #20516 
Fixes #20517 
Fixes #18960
commit 2abdc19c (2020-07-25): libc: newlib: libc-hooks: Fix Coverity warning

Suppress the coverity warning on using the semaphore as
this semaphore is used and freed only in this function.

Fixes: #18960
closed 2020-07-25 by nashif