zephyrproject-rtos/zephyr #18963: [Coverity CID :203910]Memory - corruptions in /tests/bluetooth/uuid/src/main.c

aasthagr, 2019-09-06:
Static code scan issues seen in File: /tests/bluetooth/uuid/src/main.c
 Category: Memory - corruptions
 Function: test_uuid_create
 Component: Tests
 CID: 203910
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-09-06:
```
*** CID 203910:    (OVERRUN)
/tests/bluetooth/uuid/src/main.c: 83 in test_uuid_create()
77     
78     	/* Compare UUID 16 bits */
79     	zassert_false(bt_uuid_cmp(&u.uuid, BT_UUID_DECLARE_16(0x0001)) == 0,
80     		     "Test UUIDs match");
81     
82     	/* Compare UUID 128 bits */
>>>     CID 203910:    (OVERRUN)
>>>     Overrunning struct type bt_uuid of 1 bytes by passing it to a function which accesses it at byte offset 16.
83     	zassert_false(bt_uuid_cmp(&u.uuid, &le_128.uuid) == 0,
84     		     "Test UUIDs match");
85     
86     	/* Compare swapped UUID 16 bits */
87     	zassert_true(bt_uuid_cmp(&u.uuid, BT_UUID_DECLARE_16(0x0100)) == 0,
88     		     "Test UUIDs don't match");
/tests/bluetooth/uuid/src/main.c: 67 in test_uuid_create()
61     
62     	/* Compare UUID 16 bits */
63     	zassert_true(bt_uuid_cmp(&u.uuid, BT_UUID_DECLARE_16(0x0001)) == 0,
64     		     "Test UUIDs don't match");
65     
66     	/* Compare UUID 128 bits */
>>>     CID 203910:    (OVERRUN)
>>>     Overrunning struct type bt_uuid of 1 bytes by passing it to a function which accesses it at byte offset 16.
67     	zassert_true(bt_uuid_cmp(&u.uuid, &le_128.uuid) == 0,
68     		     "Test UUIDs don't match");
69     
70     	/* Compare swapped UUID 16 bits */
71     	zassert_false(bt_uuid_cmp(&u.uuid, BT_UUID_DECLARE_16(0x0100)) == 0,
72     		     "Test UUIDs match");
```

jhedberg, 2019-09-06:
Marked as false positive in Coverity

closed 2019-09-06 by jhedberg