zephyrproject-rtos/zephyr #18964: [Coverity CID :203911]Memory - corruptions in /tests/bluetooth/uuid/src/main.c

aasthagr, 2019-09-06:
Static code scan issues seen in File: /tests/bluetooth/uuid/src/main.c
 Category: Memory - corruptions
 Function: test_uuid_create
 Component: Tests
 CID: 203911
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-09-06:
```
*** CID 203911:    (OVERRUN)
/tests/bluetooth/uuid/src/main.c: 63 in test_uuid_create()
57     
58     	/* Create UUID from LE 16 bit byte array */
59     	zassert_true(bt_uuid_create(&u.uuid, le16, sizeof(le16)),
60     		     "Unable create UUID");
61     
62     	/* Compare UUID 16 bits */
>>>     CID 203911:    (OVERRUN)
>>>     Overrunning struct type bt_uuid_16 of 4 bytes by passing it to a function which accesses it at byte offset 16.
63     	zassert_true(bt_uuid_cmp(&u.uuid, BT_UUID_DECLARE_16(0x0001)) == 0,
64     		     "Test UUIDs don't match");
65     
66     	/* Compare UUID 128 bits */
67     	zassert_true(bt_uuid_cmp(&u.uuid, &le_128.uuid) == 0,
68     		     "Test UUIDs don't match");
/tests/bluetooth/uuid/src/main.c: 71 in test_uuid_create()
65     
66     	/* Compare UUID 128 bits */
67     	zassert_true(bt_uuid_cmp(&u.uuid, &le_128.uuid) == 0,
68     		     "Test UUIDs don't match");
69     
70     	/* Compare swapped UUID 16 bits */
>>>     CID 203911:    (OVERRUN)
>>>     Overrunning struct type bt_uuid_16 of 4 bytes by passing it to a function which accesses it at byte offset 16.
71     	zassert_false(bt_uuid_cmp(&u.uuid, BT_UUID_DECLARE_16(0x0100)) == 0,
72     		     "Test UUIDs match");
73     
74     	/* Create UUID from BE 16 bit byte array */
75     	zassert_true(bt_uuid_create(&u.uuid, be16, sizeof(be16)),
76     		     "Unable create UUID");
/tests/bluetooth/uuid/src/main.c: 87 in test_uuid_create()
81     
82     	/* Compare UUID 128 bits */
83     	zassert_false(bt_uuid_cmp(&u.uuid, &le_128.uuid) == 0,
84     		     "Test UUIDs match");
85     
86     	/* Compare swapped UUID 16 bits */
>>>     CID 203911:    (OVERRUN)
>>>     Overrunning struct type bt_uuid_16 of 4 bytes by passing it to a function which accesses it at byte offset 16.
87     	zassert_true(bt_uuid_cmp(&u.uuid, BT_UUID_DECLARE_16(0x0100)) == 0,
88     		     "Test UUIDs don't match");
89     }
90     
91     /*test case main entry*/
92     void test_main(void)
/tests/bluetooth/uuid/src/main.c: 79 in test_uuid_create()
73     
74     	/* Create UUID from BE 16 bit byte array */
75     	zassert_true(bt_uuid_create(&u.uuid, be16, sizeof(be16)),
76     		     "Unable create UUID");
77     
78     	/* Compare UUID 16 bits */
>>>     CID 203911:    (OVERRUN)
>>>     Overrunning struct type bt_uuid_16 of 4 bytes by passing it to a function which accesses it at byte offset 16.
79     	zassert_false(bt_uuid_cmp(&u.uuid, BT_UUID_DECLARE_16(0x0001)) == 0,
80     		     "Test UUIDs match");
81     
82     	/* Compare UUID 128 bits */
83     	zassert_false(bt_uuid_cmp(&u.uuid, &le_128.uuid) == 0,
84     		     "Test UUIDs match");
```

jhedberg, 2019-09-06:
Marked as false positive in Coverity

closed 2019-09-06 by jhedberg