zephyrproject-rtos/zephyr #18973: The z_arch_system_halt() does not block interrupts

pizi-nordic, 2019-09-06:
**Describe the bug**
The `z_arch_system_halt()` is implemented in the following way:
```
FUNC_NORETURN __weak void z_arch_system_halt(unsigned int reason)
{
        ARG_UNUSED(reason);

        /* TODO: What's the best way to totally halt the system if SMP
         * is enabled?
         */

        (void)z_arch_irq_lock();
        for (;;) {
                k_cpu_idle();
        }
}
```

This function seems to block all interrupts and halt the CPU, which is intention here.
However the `k_cpu_idle()` unconditionally enable interrupts. As result, the interrupts
are not blocked here.

**Expected behavior**
The ` z_arch_system_halt()` should completely halt the system.

pizi-nordic, 2019-09-06:
FYI: @andyross, @ioannisg

carlescufi, 2019-09-06:
I think I've seen this during my debugging sessions actually! This matches my observations.

ioannisg, 2019-09-06:
is this a bug? Do we document that z_arch_system_halt needs to halt execution completely, or just the "kernel"?

pizi-nordic, 2019-09-06:
The interrupts are part of the "kernel", so not even kernel is halted...

ioannisg, 2019-09-06:
@pizi-nordic can a timer interrupt trigger a context-switch if we are already in z_arch_system_halt()?

andrewboie, 2019-09-06:
I sent a PR to fix this

andrewboie, 2019-09-06:
> @pizi-nordic can a timer interrupt trigger a context-switch if we are already in z_arch_system_halt()?

Yes

ioannisg, 2019-09-06:
> 
> 
> > @pizi-nordic can a timer interrupt trigger a context-switch if we are already in z_arch_system_halt()?
> 
> Yes

Then this is a bug

andyross, 2019-09-06:
FWIW: the misfeature where `k_cpu_idle()` is expected to be called with an interrupt lock and to release it before returning is an API bug.  Obviously it needs to enable interrupts internally (so that something can wake it up), but the interrupt state really ought to be consistent on return.  Obviously fixing that involves an incompatible change to our architecture API.  But we really should do it...

PR #18977 (2019-09-07): kernel: fix default z_arch_cpu_halt()
k_cpu_idle() re-enables interrupts. Just spin
instead.

Fixes: #18973
commit 90e65360 (2019-09-07): kernel: fix default z_arch_cpu_halt()

k_cpu_idle() re-enables interrupts. Just spin
instead.

Fixes: #18973
closed 2019-09-07 by nashif