zephyrproject-rtos/zephyr #18986: DTS: transition from alias to node label as the standard prefix

pabigot, 2019-09-06:
From https://github.com/zephyrproject-rtos/zephyr/pull/18937#issuecomment-528849556:

Currently the only instance-agnostic reference to a specific devicetree node is one generated from an alias to the node's label. Take this example:

```
#define DT_NORDIC_NRF_SAADC_40007000_BASE_ADDRESS   0x40007000
#define DT_ALIAS_ADC_0_BASE_ADDRESS                 DT_NORDIC_NRF_SAADC_40007000_BASE_ADDRESS
#define DT_NORDIC_NRF_SAADC_ADC_0_BASE_ADDRESS      DT_NORDIC_NRF_SAADC_40007000_BASE_ADDRESS
#define DT_INST_0_NORDIC_NRF_SAADC_BASE_ADDRESS     DT_NORDIC_NRF_SAADC_40007000_BASE_ADDRESS
```
is produced from:

                adc: saadc@40007000 {
                        compatible = "nordic,nrf-saadc";
                        reg = <0x40007000 0x1000>;
                        interrupts = <7 1>;
                        status = "disabled";
                        label = "ADC_0";
                };

plus:

        aliases {
                adc-0 = &adc;
        };

Creating an alias `adc-0` just to identify the node that already has label `adc` makes for ugly devicetree sources.  We should instead be able to eliminate the alias, and get something like:

```
#define DT_NORDIC_NRF_SAADC_40007000_BASE_ADDRESS   0x40007000
#define DT_LABEL_ADC_BASE_ADDRESS                   DT_NORDIC_NRF_SAADC_40007000_BASE_ADDRESS
#define DT_INST_0_NORDIC_NRF_SAADC_BASE_ADDRESS     DT_NORDIC_NRF_SAADC_40007000_BASE_ADDRESS
```

Here the label from the node allows code to reference using the label, rather than through an alias that adds no value.  That doesn't mean we can't keep the `DT_ALIAS_` prefix just as it is when there is an alias, but the label on the node is generally all we need.

I can't tell whether `DT_NORDIC_NRF_SAADC_ADC_0_BASE_ADDRESS` came from the alias `adc-0` or the label property `"ADC_0"`, but in either case it should be unnecessary too.

From a slack conversation discussing how to identify whether devices required by a particular sample or test are available, it was proposed to use:
```
dt_depends: dt_alias_label("pwm-0") or dt_alias_label("pwm-1") or dt_alias_label("pwm-2") or dt_alias_label("pwm-3")
```

With the change proposed here it would be more clear to use `dt_has_alias("pwm-0")` and `dt_has_label("pwm0")` since the two differ in what prefix required when accessing them through code.

pabigot, 2019-09-06:
cc @anangl

ulfalizer, 2019-09-08:
The thinking behind using `aliases` is probably that it matches how devicetree is used in Linux and elsewhere. Labels disappear from the compiled devicetree when using `dtc`, so you wouldn't be able to reference labels there. In the compiled `dtc` output, you get the equivalent of

```
aliases {
    adc-0 = "/some/path/saadc@40007000";
};
```

(Can see it by running `dtc -Odts` on `.dts` files too.)

IMO, having some level of compatibility for `.dts` files might be nice (even if it's not 100%). Making label names significant for output might come across as weird.

Note that you can put any number of labels on a node btw, like

```
not: sure: how: to: deal: with: this: some-node {
    ...
};
```

CC @b0661

pabigot, 2019-09-08:
> The thinking behind using `aliases` is probably that it matches how devicetree is used in Linux and elsewhere.

I think that's a misinterpretation of what Linux does.

`alias` appears to be relatively rare in Linux devicetree sources, as one would expect.  Most phandle references in Linux devicetree sources are to the node label.  This issue suggests we follow that standard.

The compiled form discards the labels because it gives you the entire tree of nested nodes with their node names.  In that situation you look at `/aliases/adc-0` to get `/some/path/saadc@40007000`,  then walk down the tree following that path to get the corresponding properties.  

Zephyr's tooling throws away the path and node names, flattening the structure: we can't walk a path of joined node names.  The label, which has to be unique within the source, is sufficient to uniquely identify the node in Zephyr.  We don't need an alias when we have a label, as long as the tooling provides the information associated with the labelled node as suggested above.  Adding an alias to the label is an unnecessary level of indirection.

You need an alias only when you need a generic name that might be referenced from overlays or other things that can't directly access a node.  This is exactly what we do for things like `led-0`, and why we still need `DT_ALIAS_FOO`--but only for those rare cases.

> ```
> not: sure: how: to: deal: with: this: some-node {
>     ...
> };
> ```

Deal with it by adding multiple labels to each node.

You can also attach labels to properties, including individual elements within a phandle array.  Now *that* I don't know how you're going to deal with.

ulfalizer, 2019-09-08:
> `alias` appears to be relatively rare in Linux devicetree sources, as one would expect. Most phandle references in Linux devicetree sources are to the node label. This issue suggests we follow that standard.

Note that `adc-0 = &adc;` is a path reference and not a phandle reference (does not appear within `<>`). It expands to a string with the path to the node in it, instead of to a phandle (which is just an integer internally).

`aliases` seems common in Linux too:

```sh
$ git grep 'aliases {' | wc -l
946
```

Could generate stuff from the node name as well, when it makes sense. It's using the label strings that feels a bit iffy to me, because that's information that's lost with the "standard" tools.

> You can also attach labels to properties, including individual elements within a phandle array. Now _that_ I don't know how you're going to deal with.

There's already support for it in dtlib at least (they end up in `Property.labels`). You can also put labels within property values, and those end up in `Property.offset_labels`.

`dtlib.py` supports pretty much everything `dtc` supports (except for some obscure stuff like `/plugin/` and pre-`/dts-v1/` files). I went through the C sources and implemented all the stuff they implemented.

ulfalizer, 2019-09-08:
Might help to see an example of how it works in `dtc` btw.

This file,

```
/dts-v1/;

/ {
	foo {
		phandle-ref = <&label>;
		phandle-ref-2 = <&{/bar}>;
		path-ref = &label;
	};
	label: bar {
	};
};
```

is turned into this (by `dtc -Odts foo.dts`):

```
/dts-v1/;

/ {

	foo {
		phandle-ref = < 0x01 >;
		phandle-ref-2 = < 0x01 >;
		path-ref = "/bar";
	};

	label: bar {
		phandle = < 0x01 >;
	};
};
```

That output matches the binary output format as well, except the binary output format does not preserve the label (can try `dtc foo.dts | grep label`).

pabigot, 2019-09-08:
> `aliases` seems common in Linux too:

Yes, about 30% of the 3000+ devicetree source files in Linux have an aliases node.   If you look at them, though, the number of aliases in each is significantly smaller than we're currently forced to use, and they're generic names like `rtc0`, not names specific to the device as in #18937.

> Could generate stuff from the node name as well, when it makes sense.

The node name is unique only within the node, so that won't work.  You could generate strings that express the entire path to the node, but they'd be too long.

>  It's using the label strings that feels a bit iffy to me, because that's information that's lost with the "standard" tools.

As you noted it's only lost in the binary representation.  If Zephyr retained the structure and node paths of the devicetree we wouldn't need it either.  We have to be able to access properties of specific nodes, and we can't use paths.  Using the label solves the problem.

ulfalizer, 2019-09-08:
Dunno if the `aliases` solution is thaaat bad though. It's a mechanism that's already there for referring to specific nodes with a different name. Feels cleaner than starting to do magic things with label names, even if we get a few more aliases to work around lost path information.

I'd rather see some other kind of reorganization I think, and in the meantime, it's not horrible I think.

ulfalizer, 2019-09-08:
(What's kinda horrible is having most of this stuff undocumented though. Will try to do something about that a bit later.)

erwango, 2019-09-09:
I do agree with @pabigot here. 
Interests of `alias` are to provide genericity to boards specific devices so they could be used with zephyr common pieces, like for instance:
- to introduce generic device references that could be used to stitch various boards dts with a generic overlay (things like `arduino_spi`)
- to produce a zephyr generic defines that could be used in generic zephyr applications (`DT_ALIAS_LED0_GPIOS_CONTROLLER`)

Using `alias` inside a board's dts is purely a trick to compensate lacks/weaknesses in the way we  generate dts stuff. I do not point finger at the scripts, I'm aware that there is no easy solution, but we should be aware this kind of usage of `alias` should eventually go away.

galak, 2019-09-10:
I feel like this isn't addressing the root of the issue which is the desire to have SoC specific naming scheme.  I'd suggested we introduce a new property that vendors can use to match their documentation ("soc-label").  This removes the need to various hacks of using aliases or node labels:

https://github.com/zephyrproject-rtos/zephyr/pull/12546

pabigot, 2019-09-10:
No; that's the immediate driver due to #18937.  The general case is to reference nodes in the devicetree.

`soc-label` is wrong, because I may be doing something like:
```
&adc {
   ext-therm: voltage-divider@0 { };
   int-therm: voltage-divider@1 { };
};
```

`ext-therm` is not a soc resource.

If you don't want to use the node label, we need another solution.  I had proposed a `zephyr,id` property long ago.  Today it seems even more appropriate, since this is solving a problem specific to Zephyr's approach to devicetree.

galak, 2019-09-10:
What's the driver code look like that would access `ext-therm` vs `int-therm`?  Is the driver doing something different based on which one it is?

pabigot, 2019-09-19:
I've been convinced by the arguments that using the node label is not in line with standard devicetree practice, so I no longer support the solution I proposed.  I'm closing this and redirecting to #19285.

closed 2019-09-19 by pabigot