zephyrproject-rtos/zephyr #18999: assignment in assert in test of arm_thread_arch causes build failures

pabigot, 2019-09-07:
https://github.com/zephyrproject-rtos/zephyr/blob/6e46a64a489d5cce2b7621c5e5852fc3e1235a38/tests/arch/arm/arm_thread_swap/src/arm_thread_arch.c#L468

is producing build failures on two of five platforms in #18991:

```
rtos/zephyr/tests/arch/arm/arm_thread_swap/src/arm_thread_arch.c: In function 'test_arm_thread_swap':
/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/tests/arch/arm/arm_thread_swap/src/arm_thread_arch.c:468:2: error: suggest parentheses around assignment used as truth value [-Werror=parentheses]
  zassert_true(_current->arch.swap_return_value = SWAP_RETVAL,
  ^~~~~~~~~~~~
```

As far as I can tell the diagnostic is correct.

Am I wrong?  If not, how did this get past release testing?

pabigot, 2019-09-07:
To answer my question, it must be being produced by some patch that was merged after release, but not rebased before it was merged.

pabigot, 2019-09-07:
Very odd: `git bisect` shows the failure started when 48f97ecf463d642297f61ac8b2ce3748469b98b4 which is #18945 was merged.

pabigot, 2019-09-07:
I'm reasonably certain this test is not correct, but it's not at all clear what it should do, and #19001 at least makes it "pass".

pabigot, 2019-09-13:
I've looked into this, and in fact it is the first commit in #18991 that produces this build error.  But it's not that commit that's wrong.

That patch changes the type of the first argument to `z_zassert`, which is a static inline function, from `int` to `bool` to avoid some C++ warnings about implicit conversion.   It also then changes a condition `if (!(cond)) {}` to `if (cond == false) {}`, which is perfectly acceptable since `cond` is a parameter to the function.

The bug here is because the call site has `zassert_true(lhs = rhs, ...)` which is an assignment, not a compare.  It turns out that when GCC inlines the argument it does analysis as though it had been expanded to `if (lhs = rhs == false) {}` which properly gets the warning above.  Semantically it does the right thing: it does _not_ reevaluate the original parameter with its side effects.

In short, the use of an `int` declaration for a boolean argument, and the structure of the conversion from the expression to a boolean value, has been hiding any existing code that incorrectly used an assignment instead of a compare in the condition given to `zassert_true()`.

pabigot, 2019-09-14:
@ioannis I modified #18991 to patch this test by converting the assignments to comparisons.  That wasn't successful:

    mps2_an385                tests/arch/arm/arm_thread_swap/arch.arm.swap.common FAILED: failed
    Assertion failed at ZEPHYR_BASE/tests/arch/arm/arm_thread_swap/src/arm_thread_arch.c:471:
       test_arm_thread_swap: (_current->arch.swap_return_value == swap_return_val is false)
       Swap value not returned as expected

This blocks #18991.

ioannisg, 2019-09-20:
Going to pick this up @pabigot . Sorry for the delay on this.

PR #19310 (2019-09-22): Test arch arm thread swap minor fix
Fixes a bug in erroneous zassert expression in arm thread swap test suite

Fixes #18999
closed 2019-09-22 by ioannisg