zephyrproject-rtos/zephyr #19003: bluetooth : Mesh: adv thread can be replace by other methods.

menglingao1997, 2019-09-08:
@jhedberg As we all know, creating a Task requires a separate stack space. I can see that the function of the `adv thread` can be replaced by the `k_delay_work` API, and it is not necessary to occupy `768` bytes of RAM.
I wrote some pseudo code to verify my thoughts, I hope you can give valuable advice.
```C

enum sys_states
{
	K_SYS_INIT,
#ifdef PROXY
	K_QUEUE_WAIT_PROXY,
#else
	K_QUEUE_WAIT_ADV,
#endif
	K_SLEEP_ADV,
};

static sys_states states = K_SYS_INIT;

enum sys_events{
#ifdef PROXY
	K_QUEUE_TIMEOUT_PROXY,
#endif
	K_SLEEP_TIMEOUT_ADV,
}

void k_queue_timeout_proxy_handler()
{
	if(states != K_QUEUE_WAIT_PROXY)
		__ASSERT("ERROR");

	proxy_stop();
	sys_start();
}

void k_sleep_timeout_adv_handler()
{
	if(states != K_SLEEP_ADV)
		__ASSERT("ERROR");

	adv_stop();
	sys_start();
}

void proxy_start()
{
	int timeout = proxy_send();
	states = K_QUEUE_WAIT_PROXY;

	k_delay_work_submit(proxy_handler, timeout);
}

void adv_send()
{
	adv_start();
	states = K_SLEEP_ADV;
	
	k_delay_work_submit(adv_handler, duration);
}

/*adv thread*/
void sys_start()
{
	/*global variable*/
	buf = k_queue_get();
#ifndef PROXY
	if(!buf)
	{
		states = K_QUEUE_WAIT_ADV;
		return;
	}
#endif
	
	if(buf)
	{
		adv_send();
		return;
	}

#ifdef PROXY
	proxy_start();
#endif
}

void proxy_stop()
{
	k_delay_work_cancel(proxy_handler);
}

void sys_k_queue_update_handler()
{
	switch (states)
	{
	case K_SYS_INIT:
		break;
#ifdef PROXY
	case K_QUEUE_WAIT_PROXY:
		proxy_stop();
		break;
#else
	case K_QUEUE_WAIT_ADV:
		break;
#endif
	case K_SLEEP_ADV:
		/*ignore update event when adv sleep.*/
		return;
	default:
		__ASSERT("ERROR");
	}

	sys_start();
}

void MESHAdv_Init()
{
	k_delay_work_init(proxy_handler, k_queue_timeout_proxy_handler);
	k_delay_work_init(adv_handler, k_sleep_timeout_adv_handler);
}
```

menglingao1997, 2019-09-08:
ooh, no need maintain much more k_delayed_work instance, just only one k_delay_work instance.

menglingao1997, 2019-09-08:
```C

typedef enum sys_states
{
	K_SYS_INIT,
#ifdef PROXY
	K_QUEUE_WAIT_PROXY,
#else
	K_QUEUE_WAIT_ADV,
#endif
	K_SLEEP_ADV,
};

static sys_states states = K_SYS_INIT;

static void k_adv_handler()
{
	switch (states)
	{
	case K_SYS_INIT:
		break;
#ifdef PROXY
	case K_QUEUE_WAIT_PROXY:
		proxy_stop();
		break;
#else
	case K_QUEUE_WAIT_ADV:
		break;
#endif
	case K_SLEEP_ADV:
		adv_stop();
	default:
		__ASSERT("ERROR");
	}
	
	sys_start();
}

static void proxy_start()
{
	int timeout = proxy_send();
	states = K_QUEUE_WAIT_PROXY;

	k_delay_work_submit(proxy_handler, timeout);
}

static void adv_send()
{
	adv_start();
	states = K_SLEEP_ADV;
	
	k_delay_work_submit(adv_handler, duration);
}

static void k_adv_thread()
{
	/*global variable*/
	buf = sys_slist_get(adv_queue);
#ifndef PROXY
	if(!buf)
	{
		states = K_QUEUE_WAIT_ADV;
		return;
	}
#endif
	
	if(buf)
	{
		adv_send();
		return;
	}

#ifdef PROXY
	proxy_start();
#endif
}

static void proxy_stop()
{
	k_delay_work_cancel(proxy_handler);
	proxy_adv_stop();
}

void bt_mesh_adv_update()
{
	/*only flush queue when state not in adv sleep mode.*/
	if(states != K_SLEEP_ADV){

		/*we shuld cancel the delay work.*/
		k_delay_work_cancel(adv_update);
		k_work_submit(adv_update.work);
	}
}

void bt_mesh_adv_send(struct net_buf *buf, const struct bt_mesh_send_cb *cb,
		      void *cb_data)
{
	sys_slist_put(adv_queue, net_buf_ref(buf));
	bt_mesh_adv_update();
}

void bt_mesh_adv_init()
{
	sys_slist_init(adv_queue);
	k_delay_work_init(adv_handler, k_adv_handler);
}
```

jhedberg, 2019-09-08:
@menglingao1997 I'd need to think about this a bit. To be honest, the adv thread is the last one I'd have started looking for optimisations from, since the logic of handling both proxy and network advertising can be fairly nicely implemented with a dedicated thread.

It might make sense to wait until we have the Mesh HCI vendor extensions in place to see how the code will look then, since the logic in adv.c should get simplified considerably, and this in itself might already remove the need for a dedicated thread.

menglingao1997, 2019-09-08:
I just successfully ported my idea to `nimble` just because `linux_blemesh` example can be more easily debugged on my Linux system. The purpose of my doing this is to minimize the RAM usage. At the same time, we look forward to your better solution.

vikrant8051, 2019-11-18:
@jhedberg 
Please throw some light on "Mesh vendor HCI extensions".
How it is different from current implementation ?
Pros & cons over current implementations.

Thank you !!

LingaoM, 2020-12-31:
Since @trond-snekvik solves this problem better in PR(#30005), this issure should be closed.

closed 2020-12-31 by LingaoM