zephyrproject-rtos/zephyr #19004: problems in sanitycheck/CI infrastructure revealed by post-release change

pabigot, 2019-09-08:
This summarizes some less obvious findings from investigation related to the change that was reverted in #18991.  Some of this is also in #19001.

The reversion was requested offline by @nashif due to noticing some qemu failures in some shippable results.  Independently, I noticed it in a build session where it flashed by after a config change and went away after removing `build` and reconfiguring.  Then it showed up again when an unrelated change #18999 caused a reproducible failure in an unrelated ARM thread swap test.

In both contexts most tests in CI appeared to pass, but they may not have been doing what they were supposed to do.

I ran a sanitycheck on master at 6e46a64a489d5cce2b7621c5e5852fc3e1235a38 in my development build environment and saw large numbers of failures: 298 out of 1890 tests at the point I stopped it.  That's 15% failure rate, all due to the check in CMake.

So something in the sanitycheck code may be CMake cache material somewhere that subsequent builds can pick them up, causing the check added in #18945 to generate a failure.

This failure either doesn't occur in shippable or is masked (perhaps by the matrixbuild approach) and can allow the test to run with an unusual/inconsistent configuration.  The evidence for that is in the [logs](https://app.shippable.com/github/zephyrproject-rtos/zephyr/runs/49969/2/console) from #18999 where a test build failed because conditionally-included code that produced an error diagnostic was being parsed, when normally it's excluded.  ccache may be involved.

It would be good to have an explanation for what happened.  The fact that a test that normally passed failed because of a crossed configuration suggests that tests that are passing may also be using the wrong configuration.  

The description in #19001 provides a simple reproducing case for the sanitycheck situation.

nashif, 2019-09-17:
This is not a sanitycheck issue, try:

```
gale:zephyr(cmake_blah): west build -b qemu_x86 samples/hello_world/
-- west build: making build dir /home/anashif/zephyrproject/zephyr/build pristine
-- west build: build configuration:
       source directory: /home/anashif/zephyrproject/zephyr/samples/hello_world
       build directory: /home/anashif/zephyrproject/zephyr/build
       BOARD: qemu_x86 (origin: command line)
-- west build: generating a build system
Zephyr version: 2.0.99
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.8", minimum required is "3.4")
-- Selected BOARD qemu_x86
-- Found west: /home/anashif/.local/bin/west (found suitable version "0.6.0", minimum required is "0.6.0")
-- Loading /home/anashif/zephyrproject/zephyr/boards/x86/qemu_x86/qemu_x86.dts as base
-- Overlaying /home/anashif/zephyrproject/zephyr/dts/common/common.dts
Device tree configuration written to /home/anashif/zephyrproject/zephyr/build/zephyr/include/generated/generated_dts_board.conf
Parsing Kconfig tree in /home/anashif/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/anashif/zephyrproject/zephyr/boards/x86/qemu_x86/qemu_x86_defconfig'
Merged configuration '/home/anashif/zephyrproject/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/home/anashif/zephyrproject/zephyr/build/zephyr/.config'
-- Cache files will be written to: /home/anashif/.cache/zephyr
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk-0.10.3/i586-zephyr-elf/bin/i586-zephyr-elf-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/anashif/zephyrproject/zephyr/build
-- west build: building application
[1/103] Preparing syscall dependency handling

[96/103] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
             ROM:       15788 B         4 MB      0.38%
             RAM:       65708 B         4 MB      1.57%
        IDT_LIST:         416 B         2 KB     20.31%
[103/103] Linking C executable zephyr/zephyr.elf
gale:zephyr(cmake_blah): west  build -c
-- west build: build configuration:
       source directory: /home/anashif/zephyrproject/zephyr/samples/hello_world
       build directory: /home/anashif/zephyrproject/zephyr/build
       BOARD: qemu_x86 (origin: CMakeCache.txt)
-- west build: generating a build system
CMake Error at /home/anashif/zephyrproject/zephyr/cmake/app/boilerplate.cmake:34 (message):
  'project()' was called before boilerplate.cmake was included, this is not
  supported.
Call Stack (most recent call first):
  CMakeLists.txt:5 (include)

-- Configuring incomplete, errors occurred!
See also "/home/anashif/zephyrproject/zephyr/build/CMakeFiles/CMakeOutput.log".
See also "/home/anashif/zephyrproject/zephyr/build/CMakeFiles/CMakeError.log".
ERROR: command exited with status 1: /usr/local/bin/cmake -B/home/anashif/zephyrproject/zephyr/build -S/home/anashif/zephyrproject/zephyr/samples/hello_world -GNinja
```

pabigot, 2019-09-17:
@nashif Your example does not fail on current master at af8df4f764b9348981badf6a9c8f952b043582bd; it only fails in the way you show if #18945 is restored.

I believe @galak believes the underlying problem is due to cmake errors not causing the build to fail, which is a newly discovered problem with a fix in the works somewhere.

nashif, 2019-09-17:
> @nashif Your example does not fail on current master at af8df4f; it only fails in the way you show if #18945 is restored.
 
Correct, this fails with 6e46a64 

> I believe @galak believes the underlying problem is due to cmake errors not causing the build to fail, which is a newly discovered problem with a fix in the works somewhere.

not sure I understand the above. In any case, this really is a build system issue, not related to sanitycheck.

SebastianBoe, 2019-10-15:
@pabigot : I need help understanding the problem.

Would you be able to explain the bug by listing a set of reproducable steps and a resulting expected behaviour vs. actual behaviour ?

pabigot, 2019-10-15:
See:
> The description in #19001 provides a simple reproducing case for the sanitycheck situation.

You would have to reset your HEAD to 6e46a64a489d5cce2b7621c5e5852fc3e1235a38.  Since the world has changed since I filed this, feel free to close this.

SebastianBoe, 2019-10-15:
I see that  6e46a64 is affected by the bug introduced by the "project check". I'm unable to explain how sanitycheck is able to provoke this bug, because it requires running CMake twice, but I'm hoping that if there is something other than the "project check" that is causing problems then we will see it again some day.

Closing.

closed 2019-10-15 by SebastianBoe