zephyrproject-rtos/zephyr #19012: Enable Address Sanitizer (ASAN) during sanitycheck

vanwinkeljan, 2019-09-08:
**Is your enhancement proposal related to a problem? Please describe.**

In case sanitycheck is run with address sanitizer (ASAN) enabled failures are detected.

**Describe the solution you'd like**

Enable address sanitizer (`CONFIG_ASAN`) by default for native_posix and native_posix64 when running sanitycheck.

**Describe alternatives you've considered**

Periodically (nightly, weekly,...) run sanitycheck with ASAN enabled

**Additional context**

List of PRs fixing failures detected by ASAN
* #18631
* #18633
* #18636
* #18637
* #19008
* #19010
* #19011

aescolar, 2019-09-09:
This should be perfectly fine, as even though we will increase a bit the native_posix runtime, it is very small today in CI
Care to send a PR?

vanwinkeljan, 2019-09-09:
@aescolar OK I will send out a PR once the dependencies have been merged.

In the meantime people have time to post any concerns here.

aescolar, 2019-11-02:
> This should be perfectly fine, ...

I must apologize for not having realized how much controversy and work there was going to be from this and having encouraged you to go this path.

vanwinkeljan, 2019-11-02:
> I must apologize for not having realized how much controversy and work there was going to be from this and having encouraged you to go this path.

No problem, looks like I opened pandora's box :)

closed 2024-02-09 by aescolar