zephyrproject-rtos/zephyr #19030: Extending timeouts to support multiple clocks

pabigot, 2019-09-09:
Based on discussion in #17162 the following describes a proposed API extension that is enabled by #17155 given a few minor API changes.

## Proposed Zephyr Clocks

I would like the end goal to be support for the following clocks:
* `K_CLOCK_HARDWARE` corresponds to the Zephyr hardware clock.  It is provided by `k_cycle_get()`.  This is a monotonic non-decreasing non-wrapping 64-bit source with zero at the instant the system hardware clock is initialized.  A duration of 1 corresponds to a Zephyr `cycle`.  The rate `Z_HZ_cyc` is a constant number of nominal cycles per second, but the actual rate may vary, and may not be a compile-time constant.
* `K_CLOCK_SYSTEM` corresponds to the the system clock.  This is a monotonic non-decreasing non-wrapping 64-bit source with a zero at the instant of the zero of `K_CLOCK_HARDWARE`.  A duration of 1 corresponds to a Zephyr `tick`.  The rate `Z_HZ_ticks` is a compile-time constant, but the criteria for advancing the clock depends on `CONFIG_TICKLESS_KERNEL`. For a tickless kernel this source is a linear transformation scaling `K_CLOCK_HARDWARE` by `Z_HZ_ticks / Z_HZ_cyc`.
* `K_CLOCK_MONOTONIC` is the documented clock used for the timeout API defined in #17155.  It should be an alias for `K_CLOCK_SYSTEM`.
* (TBD) `K_CLOCK_UPTIME` corresponds to `k_uptime_get()`.  This is a linear transformation using `K_CLOCK_MONOTONIC` nominal tick rate to convert to a time scale where a duration of 1 corresponds to one millisecond.  This probably isn't necessary, though it might be convenient.
* (TBD) `K_CLOCK_REALTIME` would correspond to something that tracks civil time a la NTP.  This is very TBD.

Let `k_clock_id_t` denote the type of the above `K_CLOCK_FOO` constants.  Additional clock sources are possible by a Kconfig option to enable them for specific devices.  This would extend the device API with functions provide frequency, gettime, and alarm capabilities for the provider.  For example:

    k_clock_id_t hires = counter_get_clock(dev);

would provide a clock id for timeouts that use that counter to control their duration, assuming the counter identified by `dev` was configured to support use as a clock source.  Other non-counter providers may exist as well.  Any clock id could be used in the following generic API:

    u64_t k_clock_get(k_clock_id_t id);
    u32_t k_clock_frequency(k_clock_id_t id);
    k_ticks_t k_clock_forever(k_clock_id_t id);
    k_ticks_t k_clock_convert(k_ticks_t t, k_clock_id id);

We then have:

* `k_clock_get(K_CLOCK_HARDWARE)` == `k_cycle_get()`
* `k_clock_get(K_CLOCK_UPTIME)` == `k_uptime_get()`
* `k_clock_get(K_CLOCK_SYSTEM)` returns the tick counter (`curr_tick`)

Currently the opaque `k_ticks_t` value encodes a count in ticks that is either relative or absolute.  In the future it should also encode the id of the clock to which it applies, so we should be able to use `z_add_timeout` and have that record on the queue for the corresponding clock.

We will also need API extensions:

* Every `K_TIMEOUT_suffix(args...)` will have a corresponding `K_CLOCK_TIMEOUT_suffix(id, args...)` which implements it for clocks other than `K_CLOCK_MONOTONIC`.
* All `k_?s_to_ticks_suffix(args...)` will have a corresponding `k_?s_to_clock_ticks_suffix(id, args...)` which implements the conversion between time and clock values/durations using the rate for a specific clock.

jukkar, 2019-09-10:
For gPTP (generalized Precicion Time Protocol) enabled system, there is a high accuracy clock source, with nanosecond accuracy, provided by Ethernet controller. For example frdm-k64f (eth_mcux driver) and sam-e70-xplained (eth_sam_gmac driver) boards have support for gPTP.
This clock is currently accessed only by gPTP code, which uses PTP driver for access (API at `include/ptp_clock.h`). I am not sure if this clock source would need to be accessed by other system than gPTP, so just fyi atm.

pabigot, 2019-09-10:
If that clock provides high-accuracy realtime (e.g. UTC or TA1) it could certainly be useful.

pizi-nordic, 2019-09-11:
Some questions/comments:

- Are we going to have alarms set on given clock source (including `K_CLOCK_HARDWARE`) and custom-clock sources? If so, who will be responsible for dealing with low-level constraints (example: I cannot set alarm earlier that X clock from now).
- What we should do if given clock is not available (for example when system is going into low power state) but there is alarm set on that clock?
- Are we going to inform the application about "clock lost" / "clock restored" event? Or rather application should "lock" given clock source to ensure that it will be ticking all the time?
- Have you considered https://github.com/zephyrproject-rtos/zephyr/issues/14306 in the design? Ideally we should be able to 'move' wake-up event from one clock source to another.

sslupsky, 2019-11-20:
I have not read through all of the related issues but one thing appears to be absent from this discussion, which is power consumption.  I've been working with the SAMD21 and Zephyr recently and have discovered that the clock domain used for the system timers is the same clock used to clock the CPU.  That is a problem for a low power application.  To achieve low power on the SAMD21, the SAMD21 RTC needs to be clocked by the 32KHz lower power clock source.  This implies another clock domain.  Has anyone considered the implications of this with respect to this work?

pabigot, 2019-11-20:
> This implies another clock domain. Has anyone considered the implications of this with respect to this work?

Yes.  Initial thoughts, at least.  It depends on the second TBD in #19282 relating to a domain model for clocks, which considers fixed rates, monotonicity, and the effect of discontinuities when mapping between domains.

fgrandel, 2023-06-19:
@nordic-krch @nashif @cfriedt Guess this should get an RFC tag? @nashif tagging you as you maintain an RFC backlock...

fgrandel, 2023-06-19:
> For gPTP (generalized Precicion Time Protocol) enabled system, there is a high accuracy clock source [...] not sure if this clock source would need to be accessed by other system than gPTP, so just fyi atm.

I'd vote in favor. This is what Linux does under the notion of "dynamic clocks", see clock_getres(2)

fgrandel, 2023-06-19:
@nordic-krch Some of the ideas in your Enhancement/RFC already seem to be implemented by now, `zephyr/lib/posix/clock.c` allows for access to (non-settable) CLOCK_MONOTONIC and (settable) CLOCK_REALTIME, currently both based on uptime_ticks(), ie. scaled in a POSIX compliant way similarly to what you propose as K_CLOCK_UPTIME/K_CLOCK_REALTIME. In principle I'd say these clocks could also be adjusted/compensated w/o discontinuities similarly to adjtime(3).

Would it make sense to try and map constants to those defined in POSIX and/or Linux, see clock_getres(2), e.g. 
- K_CLOCK_UPTIME -> K_CLOCK_MONOTONIC;
- K_CLOCK_SYSTEM -> K_CLOCK_MONOTONIC_RAW;
- K_CLOCK_HARDWARE -> K_CLOCK_MONOTONIC_COARSE?

(respecting the definitions of those clocks of course which is not 100% what you've defined)

This would simplify adding those to Zephyr's clock_gettime() implementation in a readable and obvious way.

@cfriedt Maybe it makes sense to close this one and continue discussion in #40099 (or the other way round). It's hard to discuss HF clock extension w/o also discussing multiple clocks.

cfriedt, 2023-06-19:
> > The monotonic clock should likely not be adjusted,
> 
> @cfriedt: Then it should probably be K_CLOCK_MONOTONIC_RAW for compatibility with Linux? CLOCK_MONOTONIC is adjustable (but not settable) by default in POSIX if I'm not mistaken. See clock_getres(2): "The  CLOCK_MONOTONIC clock is [...] affected by the incremental adjustments performed by adjtime(3) and NTP."

My guess is that you're looking at Linux man pages.

Would be better to stick with POSIX via 

https://pubs.opengroup.org/onlinepubs/9699919799/

Not 100% sure ATM if adjusting monotonic is part of POSIX. We could make it a kconfig option for a deviation though, and support it anyway.

If we were to support adjustments on monotonic on the Zephyr side, it would very likely need to be done outside of the existing k_uptime_get() machinery, but at least there is no spec to restrict that.

fgrandel, 2023-06-19:
> My guess is that you're looking at Linux man pages

You're right. :-)

> If we were to support adjustments on monotonic on the Zephyr side, it would very likely need to be done outside of the existing k_uptime_get() machinery

Exactly. Any type of compensation, calendar/leap second logic, algorithms to adjust clocks without discontinuity, etc. can be kept in the RTC layer as offset + ppm drift in the same way that z_impl_clock_gettime() already does for the realtime offset. 

The layering might be:

1) **low level counter + alarm backends**
     * cycle counter: kernel.h + time_units.h
     * any counter.h-comptible driver
     * one of the existing proprietary RTC clocks (drivers/timer/nrf_rtc_timer.h, drivers/rtc/*)

2) **zero to n rtc.h-compatible RTC wrappers** (aka rtc_emul) around each of these low level backends - as you propose iterable via a separate linker section. This layer includes all the compensation (offset + ppm drift) machinery but only as a configurable "virtual" conversion layer on top of the low-level timer, not based on anything heavy-weight like mutexes or the work queue. Above all I don't see any need for duplicate timekeeping state which consumes RAM, introduces latency and the potential for race conditions and all kinds of synchronization bugs.

Generic capabilities may be implemented along the "strategy" pattern and injected into RTC implementations (via build-time config) so that they don't have to be re-implemented per counter backend:
 * a framework to have multiple timers derived from a single counter.h source up to a KConfig(urable) max number of concurrent timers for static allocation of the required resources
 * compensation algorithms to synchronize with external clocks (e.g. a strategy to measure drift against an external clock source like PTP, ProfiBus slots or wireless TDMA protocols regularly and build a mean from it that does not accumulate errors)
 * adjustment algorithms that ensure that compensation does not introduce discontinuities into the clock

3) on top of that very **thin and independent POSIX / clib wrappers** that use characteristics/capabilites exposed by rtc.h and/or "chosen" in some standard way to map the "best" rtc clock to one of the POSIX clocks (and/or non-standard Linux-like or dynamic PTP-like extension) clocks.

This last layer could fill in any gaps not configured via "chosen" by rtc.h compatible default drivers based on the kernel cycle counter that is guaranteed to be present in all Zephyr installations.

jhedberg, 2023-06-19:
> If more detailed comparison is required then s.o. from Zephyr's BLE team would have to chime into this discussion.

@cvinayak any thoughts on the above comment from @fgrandel (https://github.com/zephyrproject-rtos/zephyr/issues/19030#issuecomment-1597226731)

fgrandel, 2023-06-30:
@cfriedt @bjarki-trackunit I quickly discussed the following questions with Bjarki over Discord but I think it makes sense to discuss them together here:

What was the rationale behind choosing an alarm (rather than timer) nomenclature and API that differs structurally from that of the system clock API? The problem I'm facing is that I'd love to use rtc.h and the emulated RTC as its default implementation in a much broader context (see my proposal above), but for that to work I have to make it tickless and sub-second.

Updates/Alarms could have been implemented using the system timer's model of relative and absolute times plus periods:
* Would reduce the learning curve as everyone could refer to their existing knowledge about the kernel system clock based timers. Choosing "alarm" rather than "timer" is rather confusing if comparing that to the kernel timer nomenclature (and in other frameworks I know of, e.g. Contiki - but it's probably a general pattern).
* Would make the zephyr,rtc-emul implementation a rather trivial and much simpler wrapper around the system clock API which it uses internally anyway.
* Would reduce the complexity of the API as the distinction between alarm and update (including different callbacks) would disappear naturally.
* Would allow for sub-second and multi-second intervals while fully supporting the original "external one-tick-per-second" RTC use case.
* Would allow for tickless timer callbacks in all cases. Waking the CPU up every second makes the current "update" interface rather problematic on battery powered devices - it defeats the purpose of a "low energy/always on RTC" in the first place.
* Timers could be allocated in caller space which would be more flexible, easier to configure and we'd get rid of the pre-allocation via devicetree altogether.

It is a bit of a pity, that the API was not introduced with the experimental tag first. Is there any way we can add this tag after the fact until we're sure the API is really mature?

bjarki-trackunit, 2023-06-30:
@fgrandel 

It is important to note that the RTC device driver API is a low level interface for the capabilities of the hardware. Real-time clocks (RTCs) have alarms, not timers nor counters, hence why the RTC API doesn't implement a timer or counter API. The reason the RTC API, which matches the capabilities of an RTC quite well, does not fit your usecase is because an RTC fundamentally does not fit your usecase.

The RTC API is not meant to abstract access to system clock sources, the functionality you are proposing here is at least one layer above device driver APIs like the RTC and Counter APIs.

Although I remain skeptical of the merit of your concerns, you are welcome to create a PR proposing the changes you would like to implement, the RTC API is experimental after all (see [API overview](https://docs.zephyrproject.org/latest/develop/api/overview.html)).

cfriedt, 2023-06-30:
@fgrandel - This API is mainly for hardware real-time clocks, and less for e.g. clock_gettime(CLOCK_REALTIME).

.. ah, I see Bjarki got to it already.

fgrandel, 2023-06-30:
> It is important to note that the RTC device driver API is a low level interface for the capabilities of the hardware.

@bjarki-trackunit @cfriedt Oh, I see your point and my fundamental misunderstanding. Thanks for clearing that up.

**TLDR;** You're right that **rtc.h would then belong to the category of "low level counter and alarm APIs"** in my layered concept above with which I'm fine. But then my argument goes the other way round as **rtc.h is still a mixture of concepts from different abstraction layers** which I propose we should keep apart in the future as proposed above.

Plus my feedback re **low-energy and locking in rtc-emul** is still valid.

I'll update my layered concept to reflect this new perspective.

**All the details follow:**

> The RTC API is inspired from [Linux RTC API](https://man7.org/linux/man-pages/man4/rtc.4.html) which is quite mature

I agree that this is a good idea - but now that I understand what rtc.h was meant for what do you think of **following Linux concepts even more closely**?

* rtc(4) is a POSIX ioctl()/poll() userland API, not a HW driver API.
* That's why Linux provides [a beautifully simple HW driver API](https://elixir.bootlin.com/linux/v6.4/source/include/linux/rtc.h#L59) a reduced version of which we could use almost unchanged as rtc.h w/o loss of functionality except for what does not belong in a HW driver anyway (see below). It is easier to implement and can be better recognized by people acquainted with Linux.
* On top of such a simple driver API we may then provide [the original POSIX API](https://elixir.bootlin.com/linux/v6.4/source/include/uapi/linux/rtc.h) iif POSIX is enabled. Linux keeps the two layers apart and Zephyr also does in most places.

Following my original layered proposal we then have the opportunity to considerably reduce the cost for RTC driver development and maintenance by **removing all higher level and non-related concepts from rtc.h which are not directly reflected in RTC hardware** in the same way as Linux does:
* **Subsecond support except for intervals is the domain of high resolution timers** that require a different API for more precision. For the same reason [rtc(4) does not have it either](https://elixir.bootlin.com/linux/v6.4/source/include/uapi/linux/rtc.h#L25) plus [uses hrtimer internally](https://elixir.bootlin.com/linux/v6.4/source/include/linux/rtc.h#L110) for subsecond periodic interrupts as optional per-instance feature iif supported by hardware.
* **Clock compensation** is not a driver feature. On Linux hwclock(8), NTP and adjtimex(2) do this - but they keep it above the driver level. On driver level it would be less confusing IMO to [tune low-level PLL parameters](https://elixir.bootlin.com/linux/v6.4/source/include/uapi/linux/rtc.h#L60) iif supported by hardware
* On Linux rtc(4) does not support **multiple alarms** at all - probably for historical reasons - but I see value in supporting it iif HW supports it natively.

In all other cases I see **more value in a higher-level re-usable real-time (optionally high-precision) multiplexed timer API** as proposed above with several HW driver APIs as backend of which rtc.h is only one. Higher-level timers with an RTC hardware driver backend can make use of HW supported alarms iif available and fall back to a software implementation otherwise plus they can build upon higher-level concepts like compensated clocks and timescales which rtc.h hardware knows nothing about.

**A full set of clock/timescale/timer features will be available in userland with much less effort for driver maintainers even if not natively supported by RTC hardware or with no RTC hardware present at all.**

Also related to multiplexed timers: If we follow my advice that alarms should only be available as supported in hardware then alarm resources can be [allocated from the stack](https://elixir.bootlin.com/linux/v6.4/source/drivers/rtc/interface.c#L826) and tracked in a [list of timers](https://elixir.bootlin.com/linux/v6.4/source/drivers/rtc/interface.c#L823) inside the timer abstraction with timer resources statically allocated in caller space to reduce complexity and resource usage.

> the RTC API is experimental after all

Erm, not really. It might be documented as such - which I have overlooked - but it is lacking the IMO much more important practical enforcement via EXPERIMENTAL flag in KConfig. We should add and backport that then, do you agree?

**Proposed action items to fix the above points:**
* **update my layered concept** with the additional information provided by @bjarki-trackunit (@fgrandel)
* **improve rtc-emul** with spinlocks, make it tickless, only update rtc_time when requested/required (@fgrandel)
* futher **discuss the architecture of the upper layer timer, clock and timescale abstractions** and how rtc.h can be stripped down accordingly (@cfriedt @bjarki-trackunit @fgrandel)
* once we agree upon an architecture: **strip down rtc.h** to a pure RTC HW driver and **move the rest to higher-layer abstractions** (@fgrandel)

bjarki-trackunit, 2023-07-01:
@fgrandel

I would suggest you focus this issue on what you actually want to achieve, instead of writing all you grievances in lists in comments. Its hard to follow and respond to concisely, leading to repetitions of answers and lack of focus.

Put simply, The RTC and Counter APIs are the lowest level APIs for the hardware (as already stated). For this reason, there is an inherent contradiction and possible lack of understanding when you state:
> strip down rtc.h to a pure RTC HW driver and move the rest to higher-layer or hardware-agnostic abstractions
This is already the case. There is nothing to strip back or move here.

Higher level abstractions are just that, higher level, built upon the lower levels.

Please read the documentation regarding API [API Status and Guidelines](https://docs.zephyrproject.org/latest/develop/api/index.html), specifically the API Lifetime section, then look at a few of the existing RTC device drivers [RTC Drivers](https://github.com/zephyrproject-rtos/zephyr/tree/main/drivers/rtc) to gain an understanding of how the API is used and what the hardware capabilities are. Lastly, look into what the rtc-emul driver is actually used for [Hint](https://github.com/zephyrproject-rtos/zephyr/tree/main/tests/drivers/rtc/rtc_api)

This should address most of your points. If you want to

fgrandel, 2023-07-02:
> I would suggest you focus this issue on what you actually want to achieve

@bjarki-trackunit You are right, the rtc.h discussion belongs in separate issues (I opened some as you've seen). Unfortunately it landed here because of my initial misunderstanding about the role of rtc.h. With hindsight the error is obvious.

fgrandel, 2024-07-26:
Closed in favor of #76335 which summarizes results from this and prior discussions for easier reference and visibility. Feel free to re-open if you think that this RFC still has merit on its own.