zephyrproject-rtos/zephyr #19031: nrfx_clock.c functions are not available with CONFIG_NRFX_CLOCK

lanvis, 2019-09-09:
**Issue**
 nrfx_clock.c functions are not defined with CONFIG_NRFX_CLOCK and therefore cannot configure RTC at run time.

**To Reproduce**
1.13.99: Add CONFIG_NRFX_CLOCK to a nrf52840 board sample project file then see if  function nrfx_clock_lfclk_stop() is available (no link error).
2.0.99 : ?? nrfx_clock.c has been removed from source tree.

**Expected behavior**
1.13.99 Functions in nrfx_clock.c are not available when CONFIG_NRFX_CLOCK is defined in proj file.
2.0.99 cmake error when defined .. found that nrfx_clock.c is no longer included as a source file and its functions (e.g. nrfx_clcock_lfclk_stop() are no where to be found).

**Impact**
Cannot configure RTC clock at run time.

**Screenshots or console output**

**Environment (please complete the following information):**
 - OS:  Linux
 - Toolchain: Zephyr SDK 0.10.3
 - Board nrf52840_pca10056
-Zephyr 1.13.99   & 2.0.99  

**Additional context**
None ...

anangl, 2019-09-10:
@lanvis Now that #18895 is merged the issue should be fixed. Could you please check and confirm it? The problem was that since the `NRFX_CLOCK` option had no prompt, it was impossible to modify it from the application configuration file - the `CONFIG_NRFX_CLOCK=y` statement was simply ignored (it was the case also in earlier versions of Zephyr, so if it worked for you earlier, it was by chance - some other Kconfig option must have selected `NRFX_CLOCK`).

lanvis, 2019-09-10:
I don't think so because in 2.0.99,  the nrfx_clock.c is not in the source tree.  This function includes all of the methods (procs) to change the RTC configuration at run time.  I looked at your commits and didn't see it.

carlescufi, 2019-09-10:
@lanvis that is because we have moved nrfx to a separate module. It's here: https://github.com/zephyrproject-rtos/hal_nordic/blob/master/nrfx/drivers/src/nrfx_clock.c

Just update to the latest master and then run `west update`. You will find the files in `modules/hal/nordic/nrfx/drivers/src`.

anangl, 2019-11-13:
Resolved by #18895.

closed 2019-11-13 by anangl