zephyrproject-rtos/zephyr #19034: sanitycheck fail with ninja option with single-core machine

ito-yu, 2019-09-10:
**Describe the bug**

- sanitycheck fail with `--ninja` option with single-core machine
- sanitycheck run completely normal with no `--ninja` option
- sanitycheck run with `--ninja` option with dual-core (or more-core) machines too.

**To Reproduce**

Run sanitycheck in single core machine with `--ninja` option.

**Expected behavior**

- sanitycheck should run even in poor spec machine 

**Impact**

In our internal test, sanitycheck run faster with `--ninja` option. So we are using `--ninja` and usually this will not cause a problem, it works on our laptops and AWS dual-core CI test server. But when sanitycheck run on single-core AWS CI machine it did fail. We want to keep same test script through the project. This issue is annoying.

**Screenshots or console output**

This is log of our gitlab ci runner.

```
$ ./zephyr/scripts/sanitycheck -v -i --ninja -T ./app/test

JOBS: 0
Reading test case configuration files under /builds/******/app/test...
Reading platform configuration files under /builds/******/zephyr/boards...
Reading platform configuration files under /builds/******/zephyr/scripts/sanity_chk/boards...
Selecting default platforms per test case
Building testcase defconfigs...
Traceback (most recent call last):
  File "./zephyr/scripts/sanitycheck", line 3631, in <module>
    main()
  File "./zephyr/scripts/sanitycheck", line 3456, in main
    discards = ts.apply_filters()
  File "./zephyr/scripts/sanitycheck", line 2240, in apply_filters
    results = mg.execute(defconfig_cb, keep_files=kept_files)
  File "./zephyr/scripts/sanitycheck", line 1454, in execute
    goal = self.goals[name]
KeyError: None
```

**Environment (please complete the following information):**

 - OS: Docker on Ubuntu Linux
 - Toolchain: Zephyr SDK 0.10.3
 - c62b57fbd4fbfc6d2e3e11fb69131e473ce76eb9

**Additional context**

Looks sanitycheck script reduce JOBS * 0.75 when we use ninja. This is ok but at least it should keep `JOBS >= 1`

galak, 2019-09-17:
We are re-working the sanitycheck script here #18603 in which this may not be an issue.  If you are willing to test that version or wait to retest.

galak, 2019-09-17:
I think this also might be fixed in 2.0.0

c97054c1c93c7c16e2af32e4aec85198da48c598

nashif, 2019-11-12:
can't reproduce, please retry with latest sanitycheck.

closed 2019-11-12 by nashif