zephyrproject-rtos/zephyr #19038: [Zephyr 1.14-stm32-netusb]:errors when i view RNDIS Device‘s properties on Windows 10

Jeepgoing, 2019-09-10:
When I test netusb sample [https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/net/sockets/echo_server](https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/net/sockets/echo_server) on STM32F4-DISCO board, some errors occur.

Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=stm32f4_disco -DOVERLAY_CONFIG=overlay-netusb.conf ..
3. make flash
4. install RNDIS driver on Windows 10
5. view properties
6. errors as follows:

> [00:00:30.869,000] <wrn> usb_rndis: Transmit response queue is not empty
> [00:00:30.869,000] <wrn> usb_rndis: Notification is already sent
> [00:00:30.869,000] <wrn> usb_rndis: Transmit response queue is not empty
> [00:00:30.869,000] <wrn> usb_rndis: Notification is already sent
> [00:00:30.870,000] <inf> usb_device: ... EP clear halt 82
> [00:00:30.870,000] <inf> usb_device: ... EP clear halt 1
> [00:00:30.871,000] <inf> usb_device: ... EP clear halt 81
> [00:00:31.059,000] <wrn> usb_rndis: Transmit response queue is not empty
> [00:00:31.059,000] <wrn> usb_rndis: Notification is already sent
> [00:00:31.067,000] <wrn> usb_rndis: Transmit response queue is not empty
> [00:00:33.475,000] <wrn> usb_rndis: Transmit response queue is not empty
> [00:00:35.979,000] <wrn> usb_rndis: Transmit response queue is not empty
> [00:00:38.480,000] <wrn> usb_rndis: Transmit response queue is not empty
> [00:00:40.983,000] <wrn> usb_rndis: Transmit response queue is not empty
> [00:00:43.487,000] <wrn> usb_rndis: Transmit response queue is not empty
> [00:00:45.991,000] <wrn> usb_rndis: Transmit response queue is not empty

Jeepgoing, 2019-09-11:
I have change `overlay-netusb.conf` as follows:

> CONFIG_USB_DEVICE_NETWORK_ECM=n
> CONFIG_USB_DEVICE_NETWORK_RNDIS=y
> CONFIG_USB_DEVICE_OS_DESC=y

Jeepgoing, 2019-09-11:
I have tested `branch:master`, it is the same error

finikorg, 2019-09-13:
Tested `echo_server` app with RNDIS and `overlay-netusb.conf` on `nucleo_f412zg` - works fine.

finikorg, 2019-09-13:
Tested `reel_board` and `frdm_k64f` works on Win10 with RNDIS.

Jeepgoing, 2019-09-16:
@finikorg  I bought a `frdm_k64f` board, tested it, works good. so i think it's STM32 usb drive proble?

finikorg, 2019-09-16:
This is somehow related to #16223. There are issue sending big packets over control endpoints for stm.

carlescufi, 2019-09-18:
@loicpoulain, @ydamigos could this be related to #16673 as well?

carlescufi, 2019-09-18:
@Jeepgoing is this reproduceable on master as well?

Jeepgoing, 2019-09-18:
@carlescufi  yeap, master can  reproduceable too

carlescufi, 2019-09-18:
> @carlescufi yeap, master can reproduceable too

Thanks, hopefully one of the maintainers can take a look.

loicpoulain, 2019-09-20:
I'm going to have a look

loicpoulain, 2019-09-20:
Tried on 96b_carbon, not getting same issue, but another one with rndis. will inverstigate.

jfischer-phytec-iot, 2019-09-25:
@loicpoulain You can use [testusb sample](https://docs.zephyrproject.org/latest/samples/subsys/usb/testusb/README.html?highlight=testusb) to reproduce the issue, please try TEST14 e.g. `testusb -v 1 -t 14 -c 3 -D /dev/bus/usb/005/056`. If I remember correctly, the device will get stuck after third transfer. Please ask me if you need help with the testusb tool.

jfischer-phytec-iot, 2019-09-25:
@loicpoulain Please take a look at #19272, there are some automatics for Setup/Data stage handling in usb_dc_stm32, so maybe it is related.

Jeepgoing, 2019-10-18:
@jfischer-phytec-iot  I tested the `branch:master`, unfortunately this question still exist

erwango, 2019-10-18:
>@jfischer-phytec-iot I tested the branch:master, unfortunately this question still exist

@Jeepgoing, can you create an issue if there is none yet?

Jeepgoing, 2019-10-18:
@jfischer-phytec-iot  I mean this issue [#19038](https://github.com/zephyrproject-rtos/zephyr/issues/19038)  has't solved.

erwango, 2019-10-18:
@Jeepgoing 

So you mean it only fails on 1.4, but working on master?
If failing on master, let's add a ticket on master as well.

Jeepgoing, 2019-11-27:
@jfischer-phytec-iot I have tested the patch #20816 it is useful, thanks.

jfischer-phytec-iot, 2019-11-27:
But it not backported yet.

Jeepgoing, 2020-04-10:
fixed

PR #21208 (2020-01-08): [Backport v1.14-branch] drivers: usb_dc_stm32: do not restrict out stage transfers to one MPS
Do not restrict control out stage transfers to one MPS.

Backport of #20816
Fixes #19038
closed 2020-04-10 by Jeepgoing