zephyrproject-rtos/zephyr #19051: [Zephyr v2.0.0 nrf52840] Unable to reconnect to recently bonded peripheral

PhilippFinke, 2019-09-10:
**Describe the bug**
I'm using the nrf52840 on my custom board as a BLE-USB-dongle. It's firmware is based on the hci-usb Zephyr sample. I've already successfully used the firmware with Zephyr v1.14.0. After upgrading to v2.0.0 I'm still able to pair and bond with devices, but after a disconnect I'm not able to successfully reconnect to the peripheral. The same issue exists with the original hci-usb sample running on a nrf52840_pca10056. We're using blueZ on a Linux machine as BT host.

**To Reproduce**
Steps to reproduce the behavior:
1. Build sample hci_usb for board nrf52840_pca10056.
2. Flash the image of hci_usb to a board.
3. Connect this board to a Linux machine using it as a BLE-dongle.
4. Build sample peripheral_sc_only for board nrf52840_pca10056.
5. Flash the image of peripheral_sc_only to another board.
6. Try to connect and pair with the hci_usb board with the peripheral_sc_only board.
7. After successful pairing try to disconnect.
8. Reconnect and wait.
9. After some time a "Connection Timeout" occurs.

**Expected behavior**
The dongle should reconnect it's bonded peripherals at any time without problems.

**Impact**
Showstopper.

**Screenshots or console output**
The first btmon console output shows the initial connection with the pairing process (creating and exchanging the keys is skipped here).

[image]

The next btmon console log shows the behaviour after a disconnect followed by another connection request. Here I'm missing the HCI Event "Encryption Change" after HCI Event "LE Start Encryption".

[image]

**Environment:**
 - OS: Linux with blueZ v5.50.
 - gnuarmemb
 - Zephyr v2.0.0

carlescufi, 2019-09-10:
@PhilippFinke can you enable the legacy cotroller and use the latest master? I think this might be related to https://github.com/zephyrproject-rtos/zephyr/issues/15256, which is fixed in the legacy controller but not yet in the split one.

carlescufi, 2019-09-16:
> @carlescufi That seems to work, but using the master branch with our product is unfortunately not an option for us. When will this be fixed in a stable release?

The fix should come as part of 1.14.1. When will that be released is still not decided but should not take long.

carlescufi, 2019-10-08:
> @carlescufi That seems to work, but using the master branch with our product is unfortunately not an option for us. When will this be fixed in a stable release?

You can wait for 2.1.0, which is now not far away. Or alternatively you can use 1.14.1 which has just been released.

joerchan, 2019-10-08:
@PhilippFinke As a workaround you could delay initiating of the security by around 15 * connection interval.

carlescufi, 2019-10-21:
@PhilippFinke you can also stay on 2.0.0 and then enable the legacy controller instead of the split one. Please ensure to do that in your `prj.conf` file. That's the options you have so far. I will close this issue and feel free to reopen or create a new one if you still experience problems.