zephyrproject-rtos/zephyr #19059: i2c_ll_stm32_v2: nack on write is not handled correctly

shphilippe, 2019-09-10:
As a master, when we do a i2c write to a device which nack during the data phase, the driver is not generating a nack correctly:

Here is an oscilloscope capture
[image]

I suspect the behavior to handle the nack in the stm32_i2c_event() is wrong. Removing the "goto end" in the if handling the nack lead to the correct result:
[image]

Before I formally submit a patch, is there some regression tests on this drivers available ?

erwango, 2019-09-13:
> @erwango Could you help us if we need to send a STOP manually if AUTOEND==0 && RELOAD == 0 && transmitted bytes < NBYTES?

Ok, will seek answer internally

erwango, 2019-09-16:
@shphilippe, @ydamigos, according to expert:
> if autoend is not enabled, it is indeed required to send a STOP manually whatever the use case

ydamigos, 2019-09-16:
@shphilippe Could you test if PR #19194 resolves the issue?

shphilippe, 2019-09-16:
@ydamigos  I'll test when I have time. But I doubt it is the correct fix, we _have_ to wait until the stop is generated. We cannot immediately free the I2C device (the goto end will do that). Thus I still think that the cleanest fix is to remove the goto end and let the I2C driver state machine generate the stop and wait until it is done to release the semaphore.

ydamigos, 2019-09-17:
@shphilippe You are right, I forgot to remove goto.

shphilippe, 2019-09-17:
@ydamigos  I'm sorry to nitpick, but the stop will be anyway generated because the TC/TCR flag is set (please @erwango  confirm datasheet is unclear, this is what I observe). Thus I think not manually forcing it in that condition is better. It is more generic since the TC/TCR handling (right below) will do the right thing if the user of the API would better try to do a repeated start instead of a stop.

ydamigos, 2019-09-17:
@shphilippe I thought the same thing at first, but I tried to keep the drivers behavior the same as before (when NACK abort transmission). What if user wants to resend a sequence of messages in case of NACK and the message, that got NACK, doesn't have STOP flag.

shphilippe, 2019-09-17:
@ydamigos  well, the driver behavior before was totally wrong. he was aborting without generating a stop. So this is definitely no something we should try to keep as a behavior.
If the user want to generate a stop, he was to put the stop flag. If he doesn't want to generate a stop and keep the bus ready for a repeated start, he should not put the stop flag.  That's what the API says. We should try to stick to this subsystem wide definition instead of having the stm32 driver defining his own. Anyway I'm not the subsystem maintainer, I'll let him make the choice.

ydamigos, 2019-09-17:
> well, the driver behavior before was totally wrong. he was aborting without generating a stop. So this is definitely no something we should try to keep as a behavior.

@shphilippe We are talking about the same thing. I am following a conservative approach here. PR fixes the wrong behavior and generates a stop before aborting.

shphilippe, 2019-09-17:
@ydamigos I fail to understand your reasoning wrt to which change is more conservative. 

Before this patch: 
- In case of before-end-of-transfer NACK: Driver does not generate STOP, current transaction is aborted, whatever the API user asked. We return -EIO to the caller. The next transaction will be a start on the bus, thus will be interpreted as a repeated start.

With your proposed change:
- In case of before-end-of-transfer NACK: Driver will force-generate a STOP, current transaction is aborted, whatever the API user asked. We return -EIO to the caller. The next transaction will be a start on the bus, thus will be interpreted as a normal start.

With the change I suggest:
- In case of before-end-of-transfer NACK: Driver will generate a stop if the user asked for it,  current transaction is aborted. We return -EIO to the caller. The next transaction will be a either a start on the bus or a repeated start, depending if the user asked to generate a STOP or not.

But I think is up to the I2C subsystem to decide which behavior the drivers should implements. Thus we have the same behavior across different i2c host easing i2c device driver portability.

ydamigos, 2019-09-18:
@shphilippe My proposed change is conservative. It fixes the issue (generates a STOP after NACK) and aborts the transmission. The task (user) should acquire the bus again to start a new transaction.

>In case of before-end-of-transfer NACK: Driver will generate a stop if the user asked for it, current transaction is aborted. We return -EIO to the caller. The next transaction will be a either a start on the bus or a repeated start, depending if the user asked to generate a STOP or not.

I don't think it is that simple. What if we are in a multi-tasking scenario and two tasks use I2C bus to talk to different devices. The both try to get the I2C bus, and the first task gets it (second task wait on bus_mutex). The first task gets a NACK in a message with no STOP flag and the second task gets the bus and creates a repeated start.

PR #19194 (2019-09-27): i2c_ll_stm32_v2: Send STOP manually after NACK
In master trasmitter mode AutoEndMode is
always disabled, so we need to send STOP
manually if NACK is received.

Fixes #19059
commit 2f4c1ea3 (2019-09-27): i2c_ll_stm32_v2: Send STOP manually after NACK

In master trasmitter mode AutoEndMode is
always disabled, so we need to send STOP
manually if NACK is received.

Fixes #19059
commit b43b5f28 (2019-09-27): i2c_ll_stm32_v2: Send STOP manually after NACK

In master trasmitter mode AutoEndMode is
always disabled, so we need to send STOP
manually if NACK is received.

Fixes #19059
commit 69725fb3 (2019-09-27): i2c_ll_stm32_v2: Send STOP manually after NACK

In master trasmitter mode AutoEndMode is
always disabled, so we need to send STOP
manually if NACK is received.

Fixes #19059
closed 2019-09-27 by MaureenHelm