zephyrproject-rtos/zephyr #19063: can we increase qemu_riscv32/64 RAM sizes

andrewboie, 2019-09-11:
Both qemu_riscv32 and qemu_riscv64 inherit from the riscv32-fe10.dtsi which provides only 16KB of RAM.

This prevents a lot of larger test cases from running on RISCV under emulation in QEMU.

Can we change this? I'd suggest a couple megabytes which would allow the largest tests to run.

I tried overriding &dtim in the board-level DTS to increase RAM size but this just caused QEMU to crash.

andrewboie, 2019-09-11:
@nategraff-sifive FYI

al3x-huang, 2019-11-03:
> All this needs to work is a patch to QEMU to increase the amount of emulated DTIM.

How does one do this

nategraff-sifive, 2019-11-04:
It just needs a patch in the [Zephyr SDK sources](https://github.com/zephyrproject-rtos/sdk-ng/tree/master/patches) which alters the size of the DTIM in [hw/riscv/sifive_e.c in the QEMU source tree.](https://git.qemu.org/?p=qemu.git;a=blob_plain;f=hw/riscv/sifive_e.c;hb=HEAD)

You'd want to edit this struct:
```
static const struct MemmapEntry {
    hwaddr base;
    hwaddr size;
} sifive_e_memmap[] = {
    ...
    [SIFIVE_E_DTIM] =     { 0x80000000,     0x4000 }
};
```

Just increase the second number to something like 0x400000

andrewboie, 2019-11-05:
There's no other way to do this other than patching the QEMU source code?

closed 2020-08-14 by andrewboie