zephyrproject-rtos/zephyr #19064: Correct docs for K_THREAD_DEFINE

gregersrygg, 2019-09-11:
**Describe the bug**
The docs states a thread can be spawned at compile time. Spawning involves starting, right? Which obviously is not possible during compilation.

> [«Alternatively, a thread can be spawned at compile time by calling K_THREAD_DEFINE»](https://github.com/zephyrproject-rtos/zephyr/blame/82c25587a2bc7610530154df4d984bdfa2add594/doc/reference/kernel/threads/index.rst#L201)

**To Reproduce**
Check the docs: https://docs.zephyrproject.org/latest/reference/kernel/threads/index.html#spawning-a-thread

**Expected behavior**
I'm not sure what `K_THREAD_DEFINE` does exactly, but I would guess it defines/declares a thread that will be spawned during system initialization. Would be nice with a clarification here.

**Impact**
Confusing documentation

**Environment (please complete the following information):**
 - OS: all
 - Toolchain: docs
 - Commit SHA: latest commit 82c25587a2bc7610530154df4d984bdfa2add594

aescolar, 2019-09-11:
CC: @andyross @andrewboie

ioannisg, 2019-10-16:
@andrewboie could we fix this for 2.1.0?

andrewboie, 2019-10-16:
I don't personally have the bandwidth to attend to this for 2.1

nashif, 2019-11-08:
this was fixed long time ago.

closed 2019-11-08 by nashif