zephyrproject-rtos/zephyr #19066: Build error with qemu_x86_64

jukkar, 2019-09-11:
**Describe the bug**
The compilation of websocket sample application in #19025 gives this build error:
```
In file included from .../zephyr/subsys/net/l2/ethernet/ethernet.c:16:
.../zephyr/include/net/gptp.h: In function 'gptp_pow2':
.../zephyr/include/net/gptp.h:72:1: error: SSE register return with SSE disabled
 {
 ^
make[3]: *** [zephyr/subsys/net/l2/ethernet/CMakeFiles/subsys__net__l2__ethernet.dir/build.make:63: zephyr/subsys/net/l2/ethernet/CMak
```

**To Reproduce**

1. Fetch #19025
2. Remove blacklisting of qemu_x86_64 from `samples/net/sockets/websocket_client/sample.yaml`
3. sanitycheck -t websocket -p qemu_x86_64

**Expected behavior**
No error

**Impact**
Test fails. I will blacklist this board temporarily in order to proceed.

gnuless, 2019-09-13:
GCC chokes on this because x86_64 lacks support for SSE and GCC doesn't know how to pass floating-point arguments (or return floating-point values) absent SSE registers.  We should probably leave the test blacklisted and close this bug.

gnuless, 2019-09-13:
On a side, note, looking at the function:

```static inline double gptp_pow2(int exp)
{
        double res;

        if (exp >= 0) {
                res = 1 << exp;
        } else {
                res = 1.0;

                while (exp++) {
                        res /= 2;
                }
        }

        return res;
}
```

Why not avoid the iteration for negative exponents, and use ``1.0 / (1 << -exp)`` ?

jukkar, 2019-09-13:
> Why not avoid the iteration for negative exponents, and use `1.0 / (1 << -exp)` ?

If the exponent is very small, then we could get wrong result.
```
orig: 2^-48 = 3.55271367880050092936e-15
  v2: 2^-48 = 1.52587890625000000000e-05
orig: 2^-96 = 1.26217744835361888866e-29
  v2: 2^-96 = 1.00000000000000000000e+00
```

gnuless, 2019-09-13:
Understood. I guess you do not expect the exponent to get very large, though. So the domain isn't symmetric about 0.  Makes enough sense given the context.

jukkar, 2019-09-16:
> I guess you do not expect the exponent to get very large, though. So the domain isn't symmetric about 0

Correct, the numbers related to gPTP are typically very small.

Lets close this then and I will leave the board blacklisted in this sample.

closed 2019-09-16 by jukkar