zephyrproject-rtos/zephyr #19075: k_delayed_work_submit() does not handle delay correctly

lemrey, 2019-09-11:
**Describe the bug**
For some large values of `delay`, `k_delayed_work_submit()` succeeds but work item is called immediately.

**To Reproduce**
Steps to reproduce the behavior:
1. Build any project for `nrf9160_pca10090`.
2. Submit a delayed work with a `delay` of `K_HOURS(20)`.
3. The work item is processed immediately.

Reproducible here: https://github.com/lemrey/zephyr/commit/1d8e92474d6001291b87e67f1e6ad10f45d0011f

**Expected behavior**
Either `k_delayed_work_submit` return an error or,
the work item submitted is handled after the specified delay.

**Additional context**
Only reproduced on Nordic hardware myself, but other hardware is similarly affected.
Actual `delay` value that overflows the tick count is dependent upon `CONFIG_SYS_CLOCK_TICKS_PER_SEC`.
Here the value given as example: `K_HOURS(20)` overflows for `CONFIG_SYS_CLOCK_TICKS_PER_SEC=32768`.

aescolar, 2019-09-11:
CC: @andyross @pabigot

lemrey, 2019-09-11:
> Even then, though, a configuration that uses 32-bit timeouts isn't going to work for durations that exceed the representable range in ticks.

That's understandable and okay, so far as the API returns an error if the current configuration does not support the requested timeout value.

pabigot, 2019-09-11:
Agreed, but with signed 32-bit delay representations the existing API was told to delay for 16:24:32 H:M:S in the past, so this is a "works as designed"-type bug.

SebastianBoe, 2019-09-13:
> Agreed, but with signed 32-bit delay representations the existing API was told to delay for 16:24:32 H:M:S in the past, so this is a "works as designed"-type bug.

Which API is "working as designed"? k_delayed_work_submit, or something under the hood?

I'm getting that k_delayed_work_submit is told to wait for K_HOURS(20) == 72000000, which is safely within the positive range of a 32 bit signed number.

pabigot, 2019-09-13:
> Which API is "working as designed"? k_delayed_work_submit, or something under the hood?

`struct _timeout`.
 
> I'm getting that k_delayed_work_submit is told to wait for K_HOURS(20) == 72000000, which is safely within the positive range of a 32 bit signed number.

If the units of the delay were milliseconds that would be correct.  They're actually ticks, which on Nordic is 2^-15 s rather than 10^-3 s.

SebastianBoe, 2019-09-13:
Ah, then we were speaking of different API's, I was speaking of the k_delayed_work_submit API, which uses ms as a unit (and is therefore not working as designed).

pabigot, 2019-09-13:
You're right; that re-exposes the problem that the kernel is not consistent, or at least forward-looking, in describing the units of the APIs that normally use the K_UNIT() time conversions, nor the implicit constraints on their values due to the varying nature of the underlying clock.  Another API change to moot before we can improve timer resolution and duration.

andyross, 2020-05-11:
We can close this one now.  Default timeout precision is now 64 bit and the docs are quite clear about what values are stored where and what the units are.

closed 2020-05-11 by andyross