zephyrproject-rtos/zephyr #19078: search for board specific shield overlays doesn't always work

galak, 2019-09-11:
If the shield name doesn't match the dir name we don't look in correct place.  For example shield `x_nucleo_iks01a3_shub`.

galak, 2019-09-11:
So we look here for things:

```
look for /home/galak/git/zephyr/boards/shields/x_nucleo_iks01a3_shub/boards/stm32mp157c_dk2.overlay
look for /home/galak/git/zephyr/boards/shields/x_nucleo_iks01a3_shub/x_nucleo_iks01a3_shub.conf
look for /home/galak/git/zephyr/boards/shields/x_nucleo_iks01a3_shub/boards/stm32mp157c_dk2.conf
```

The dir is incorrect, however not quite sure how we correctly handle this for multiple different shields in on dir like `x_nucleo_iks01a3` has.

erwango, 2019-09-11:
Shouldn't we split into 2 different shields? 
From what I understand the different modes enables quite different behavior and that's if we had 2 different shields finally.
@avisconti, what do you think ?

mike-scott, 2019-09-11:
I have to agree here.  When you enable the sensorhub mode (which is a jumper) it completely changes the functionality of the shield.

commit 2f1cd9bf (2019-10-18): cmake: add new search locations for shield overlay and conf files

* Add support to search for a board specific shield in
  <SHIELD DIR>/boards/<SHIELD>/<BOARD>.{overlay,conf}.  So we can
  support something like:

  x_nucleo_iks01a3/boards/x_nucleo_iks01a3_shub/stm32mp157c_dk2.overlay
  x_nucleo_iks01a3/boards/x_nucleo_iks01a3_shub/stm32mp157c_dk2.conf

Fixes #19078