zephyrproject-rtos/zephyr #19087: Kernel Shared Memory

bfrog, 2019-09-11:
**Is your enhancement proposal related to a problem? Please describe.**
I would like to shared memory from the kernel to user mode threads without copying, its clear
from #17194 and my own PR #17434 both share the idea that memory sharing is possible and needed to perform well.

**Describe the solution you'd like**
For my own PR #17434 it is ok if a supervisor thread sets up the shared memory space first before giving it to a user mode thread.

**Describe alternatives you've considered**
I've considered simply not supporting usermode. I've been mulling heavily the idea that
the supervisor must setup a shared memory context using the memory domain API. Though a call to free requires a pointer into the shared memory domain, others have mentioned this could cause trouble though I'm not sure how myself.

Looking at some comments made by @andrewboie regarding memory domains the limited number of them currently available might not allow using it for this purpose, if there were an unlimited number then it would be ok.

andrewboie, 2019-09-12:
@bfrog thanks for opening this.
I'm removed my -1 from your PR to un-block progress, and we can consider the best way to handle the IPC and memory sharing in this ticket. Would like to have at least a prototype for 2.1.

andrewboie, 2020-08-10:
Somewhat ties into https://github.com/zephyrproject-rtos/zephyr/issues/25950

This ticket has been dormant for a while but this is an open topic for later this year / early 2021.

teburd, 2024-02-08:
This is a relatively solved problem now, and is clearly doable with memory partitions at the cost of mpu region usage

closed 2024-02-08 by teburd