zephyrproject-rtos/zephyr #19096: No error thrown for device tree node with missing required property of type compound

ejohnso49, 2019-09-12:
**Describe the bug**
The scripts for parsing the device tree do not throw errors if a node is missing a required compound property.

**To Reproduce**
Steps to reproduce the behavior:
1. Edit a device tree node with a required compound property. From [zephyr/boards/arm/nucleo_l496zg/nucleo_l496zg.dts](https://github.com/zephyrproject-rtos/zephyr/blob/ee525c2597c78c98a3988d403d4859e4cfd78ed5/boards/arm/nucleo_l496zg/nucleo_l496zg.dts#L37)
```
pwmleds {
	compatible = "pwm-leds";
      	red_pwm_led: red_pwm_led {
		pwms = <&{/soc/timers@40014000/pwm} 1 4>;
	};
};
```
Change `pwms` to `pwm`.

2. Build the `samples/basic/fade_led` application.
3. Observe no error thrown during dts phase, but error thrown during compilation because expected defines for the node are missing.

**Expected behavior**
An error to be thrown while extracting the info from the device tree files that explains a node is missing a required property.

**Impact**
Annoyance during development as a code compilation error does not point to the direct problem in the device tree.

**Screenshots or console output**
```
(venv) $ west build -b nucleo_l496zg zephyr/samples/basic/fade_led/
-- west build: build configuration:
       source directory: /project/zephyr/samples/basic/fade_led
       build directory: /project/build (created)
       BOARD: nucleo_l496zg (origin: command line)
-- west build: generating a build system
Zephyr version: 2.0.0
-- Found PythonInterp: /project/venv/bin/python (found suitable version "3.7.4", minimum required is "3.4") 
-- Selected BOARD nucleo_l496zg
-- Found west: /project/venv/bin/west (found suitable version "0.6.2", minimum required is "0.6.0")
-- Loading /project/zephyr/boards/arm/nucleo_l496zg/nucleo_l496zg.dts as base
-- Overlaying /project/zephyr/dts/common/common.dts
Device tree configuration written to /project/build/zephyr/include/generated/generated_dts_board.conf
Parsing Kconfig tree in /project/zephyr/Kconfig
Loaded configuration '/project/zephyr/boards/arm/nucleo_l496zg/nucleo_l496zg_defconfig'
Merged configuration '/project/zephyr/samples/basic/fade_led/prj.conf'
Configuration saved to '/project/build/zephyr/.config'
-- Cache files will be written to: /Users/me/Library/Caches/zephyr
-- The C compiler identification is GNU 8.3.1
-- The CXX compiler identification is GNU 8.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /Users/me/work/toolchains/gcc-arm-none-eabi-8-2019-q3-update/bin/arm-none-eabi-gcc
-- Performing Test toolchain_is_ok
-- Performing Test toolchain_is_ok - Success
Including module: atmel in path: /project/modules/hal/atmel
Including module: civetweb in path: /project/modules/lib/civetweb
Including module: esp-idf in path: /project/modules/hal/esp-idf/zephyr
Including module: fatfs in path: /project/modules/fs/fatfs
Including module: qmsi in path: /project/modules/hal/qmsi
Including module: cypress in path: /project/modules/hal/cypress
Including module: nordic in path: /project/modules/hal/nordic
Including module: openisa in path: /project/modules/hal/openisa
Including module: microchip in path: /project/modules/hal/microchip
Including module: silabs in path: /project/modules/hal/silabs
Including module: st in path: /project/modules/hal/st
Including module: stm32 in path: /project/modules/hal/stm32
Including module: ti in path: /project/modules/hal/ti
Including module: libmetal in path: /project/modules/hal/libmetal
Including module: lvgl in path: /project/modules/lib/gui/lvgl
Including module: mbedtls in path: /project/modules/crypto/mbedtls
Including module: mcumgr in path: /project/modules/lib/mcumgr
Including module: nffs in path: /project/modules/fs/nffs
Including module: nxp in path: /project/modules/hal/nxp
Including module: open-amp in path: /project/modules/lib/open-amp
Including module: openthread in path: /project/modules/lib/openthread
Including module: segger in path: /project/modules/debug/segger
Including module: tinycbor in path: /project/modules/lib/tinycbor
Including module: littlefs in path: /project/modules/fs/littlefs
-- Configuring done
-- Generating done
-- Build files have been written to: /project/build
-- west build: building application
[1/127] Preparing syscall dependency handling

[41/127] Building C object CMakeFiles/app.dir/src/main.c.obj
FAILED: CMakeFiles/app.dir/src/main.c.obj 
ccache /Users/me/work/toolchains/gcc-arm-none-eabi-8-2019-q3-update/bin/arm-none-eabi-gcc -DBUILD_VERSION=zephyr-v2.0.0 -DCORE_CM4 -DKERNEL -DSTM32L496xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I/project/zephyr/kernel/include -I/project/zephyr/arch/arm/include -I/project/zephyr/include -I/project/zephyr/include/drivers -Izephyr/include/generated -I/project/zephyr/soc/arm/st_stm32/stm32l4 -I/project/zephyr/drivers -I/project/zephyr/ext/hal/cmsis/Include -I/project/modules/hal/stm32/stm32cube/stm32l4xx/soc -I/project/modules/hal/stm32/stm32cube/stm32l4xx/drivers/include -I/project/modules/hal/stm32/stm32cube/stm32l4xx/drivers/include/Legacy -isystem /project/zephyr/lib/libc/minimal/include -isystem /Users/me/work/toolchains/gcc-arm-none-eabi-8-2019-q3-update/bin/../lib/gcc/arm-none-eabi/8.3.1/include -isystem /Users/me/work/toolchains/gcc-arm-none-eabi-8-2019-q3-update/bin/../lib/gcc/arm-none-eabi/8.3.1/include-fixed -Os -imacros/project/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mthumb -mcpu=cortex-m4 -imacros/project/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/project/zephyr/samples/basic/fade_led=CMAKE_SOURCE_DIR -fmacro-prefix-map=/project/zephyr=ZEPHYR_BASE -ffunction-sections -fdata-sections -mabi=aapcs -march=armv7e-m -std=c99 -nostdinc -MD -MT CMakeFiles/app.dir/src/main.c.obj -MF CMakeFiles/app.dir/src/main.c.obj.d -o CMakeFiles/app.dir/src/main.c.obj   -c /project/zephyr/samples/basic/fade_led/src/main.c
/project/zephyr/samples/basic/fade_led/src/main.c:23:2: error: #error "Choose supported PWM driver"
 #error "Choose supported PWM driver"
  ^~~~~
/project/zephyr/samples/basic/fade_led/src/main.c: In function 'main':
/project/zephyr/samples/basic/fade_led/src/main.c:43:31: error: 'PWM_DRIVER' undeclared (first use in this function)
  pwm_dev = device_get_binding(PWM_DRIVER);
                               ^~~~~~~~~~
/project/zephyr/samples/basic/fade_led/src/main.c:43:31: note: each undeclared identifier is reported only once for each function it appears in
/project/zephyr/samples/basic/fade_led/src/main.c:50:33: error: 'PWM_CHANNEL' undeclared (first use in this function); did you mean 'TIM_CHANNEL_2'?
   if (pwm_pin_set_usec(pwm_dev, PWM_CHANNEL,
                                 ^~~~~~~~~~~
                                 TIM_CHANNEL_2
[50/127] Building C object zephyr/arch/arch/arm/core/CMakeFiles/arch__arm__core.dir/swap.c.obj
ninja: build stopped: subcommand failed.
ERROR: command exited with status 1: /usr/local/bin/cmake --build /project/build
```

**Environment (please complete the following information):**
 - OS: MacOS, but I think this applies to any OS
 - Toolchain: arm-gcc, but I think this applies to any toolchain
 - zephyr-v2.0.0

**Additional context**
I believe this is caused in [edt_lib.py](https://github.com/zephyrproject-rtos/zephyr/blob/7c0ec4358a205f83d2c6cc91d6dfc0a3106989a4/scripts/dts/edtlib.py#L539). Because the code returns if the `prop_type == "compound"`, there is no call to `Device._prop_val`. `Device._prop_val` has code to throw an error if a required property does not have a value. Totally understand if there's something more complicated that I'm missing though, thanks!

galak, 2019-09-12:
I think this is actually implemented post 2.0.  If you try a recent SHA I think you'll see an error message for this.

galak, 2019-09-12:
Please re-open if the current master doesn't show report an error.

closed 2019-09-12 by galak

ejohnso49, 2019-09-12:
Awesome, yep you're totally right. Post 2.0 looks good.