zephyrproject-rtos/zephyr #19098: Failed to flash on ESP32

nguyen-phuoc-dai, 2019-09-12:
I have followed the "Getting Started Guide" section to setup the environment on Windows 10 to start experience on Zephyr. Everything was OK till final step when I tried to flash zephyr to ESP32 device by "west flash -- esp-device COM44". I got "OSError: [WinError 193] %1 is not a valid Win32 application" error.

**To Reproduce**
Steps to reproduce the behavior:
1. Install windows host
2. Build sample project "hello, world" on ESP32
3. Flash the Zephyr image on real ESP32
4. Got the error

**Expected behavior**
Flash the zephyr image on the ESP32 and see "Hello, World! esp32" message on console window of putty that connected to ESP32

The following was screenshot for build and flash steps

[image]
[image]

nguyen-phuoc-dai, 2019-09-16:
Hi all,

Is anyone here see the issue? Please help tell me what i can do to flash the image on ESP32
Thanks

ExtremeGTX, 2019-09-16:
Could you please tell me if windows is 32/64-bit and if python is 32/64-bit ?
I will test this on my setup (Windows 10 64-bit/Py3.7 64-bit)

nguyen-phuoc-dai, 2019-09-17:
Hi @ExtremeGTX ,

Thanks for your support.

The following was my testing environment

Python 3.7.4
C:\Users\phuocdai>python --version
Python 3.7.4

OS was Windows 10 64bits
C:\Users\phuocdai>systeminfo

Host Name:                 DAIPC
OS Name:                   Microsoft Windows 10 Home Single Language
OS Version:                10.0.17134 N/A Build 17134
OS Manufacturer:           Microsoft Corporation
OS Configuration:          Standalone Workstation
OS Build Type:             Multiprocessor Free
Registered Owner:          HP
Registered Organization:   HP
Product ID:                00327-35818-48690-AAOEM
Original Install Date:     5/15/2018, 9:15:18 AM
System Boot Time:          9/15/2019, 10:22:55 AM
System Manufacturer:       HP
System Model:              HP ENVY Notebook 13-ab0XX
System Type:               x64-based PC

ExtremeGTX, 2019-09-17:
After some analysis i can see the problem is calling the `.py` file directly on windows
https://github.com/zephyrproject-rtos/zephyr/blob/af30f2b7d4ebb547ed9ce92494c97943e5a59331/scripts/west_commands/runners/esp32.py#L81-L82

Changing this to 
```python
cmd_convert = ['python', self.espidf, '--chip', 'esp32', 'elf2image', self.elf]
cmd_flash = ['python', self.espidf, '--chip', 'esp32', '--port', self.device,
```
makes it working.

AFAIK Windows doesn't support shebang line `#!/bin/python` so you must call the required interpreter and pass the the script as an arg.

nguyen-phuoc-dai, 2019-09-18:
Hi ExtremeGTX,

I did follow your guide and it works like a charm.

Thank you very much

ExtremeGTX, 2019-09-18:
@galak i think this needs to be reopened and fixed.

nguyen-phuoc-dai, 2019-09-18:
Hi @galak,

Please take your time to check and fix it

ioannisg, 2019-09-23:
@dcpleung could it be something you could look at?

PR #19355 (2019-09-25): west: esp32: use Python interpreter to execute ESP tool
The ESP tool is being executed directly in the esp32 runner,
assuming the tool is executable by itself. However, it would
fail under Windows as subprocess.check_call() cannot execute
Python scripts directly. The fix is to execute the Python
interpreter and passing the script path as a command line
parameter.

Fixes #19098
commit be8cb89e (2019-09-25): west: esp32: use Python interpreter to execute ESP tool

The ESP tool is being executed directly in the esp32 runner,
assuming the tool is executable by itself. However, it would
fail under Windows as subprocess.check_call() cannot execute
Python scripts directly. The fix is to execute the Python
interpreter and passing the script path as a command line
parameter.

Fixes #19098
closed 2019-09-25 by carlescufi