zephyrproject-rtos/zephyr #19133: Scheduler change in #17369 introduces crashes

olofk, 2019-09-13:
**Describe the bug**
After commit 8ec8a58fe8475d3520c376e49821d9337afa08b5 Zephyr starts crashing on our board. Reverting the two lines in that patch makes it work again

**To Reproduce**
Build instructions for the SoC here https://github.com/chipsalliance/Cores-SweRVolf

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.10.0 and 0.10.3 tested
 - Commit SHA or Version used : 8ec8a58fe8475d3520c376e49821d9337afa08b5

wentongwu, 2019-09-13:
@olofk could you please describe more for the issue? for example, your configuration, test case you are running, etc.

olofk, 2019-09-13:
My test case is a modified hello world that looks like this
#include <zephyr.h>
#include <misc/printk.h>

#include <stdio.h>

    void main(void)
    {
      int i=0;
      while(1)	printf("Hello World! %s %02x\n", CONFIG_BOARD,i++);
    }

Switching to printf and printing hex char just seems to make it more likely to crash. Perhaps because it's spending more time in the printf code. Not sure.

What I see is that UART output is garbled before it starts crashing, e.g. looking like this

    Hello World! swervolf_nexys 391
    Hello World! swervolf_nexys 392
    Hello World! swervolf_nexys 393
    Hello World! swervolf_nexys 394
    Hello World! swervolf_nexys %02x
    Hello World! swervolf_nexys 396
    Hello World! swervolf_nexys 397
    Hello World! swervolf_nexys 398
    Hello World! swervolf_nexys 399
    Hello World! swervolf_nexys 39a
    Hello World! swervolf_nexys 39b

Eventually it crashes with an unaligned memory access or instruction fault but looking at the code, there is no reasonable explanation for some of the register values which makes me suspect problems during a context switch.

The SoC itself is a riscv32 CPU running on an FPGA. In its simplest for it just has memory, UART, a single GPIO for LEDs and a riscv_machine_timer to generate timer interrupts.

There's not very much configuration applied to the soc and board ports (https://github.com/chipsalliance/Cores-SweRVolf/tree/master/zephyr). Mostly boilerplate, enabling drivers and setting the CPU frequency to 25MHz. I have tried with many combinations now of different Zephyr versions, SDK, with and without tickless_kernel, different FPGA images and with/without reverting #17369 and come to the conclusions that 1.14.0 works fine and 1.14.1-rc2 works fine after reverting #17369. zephyr 2 also crashes, but I don't know yet if that is related

wentongwu, 2019-09-13:
Ok I will consider it.
But can you please help make sure you have CONFIG_TIMESLICING enabled?

wentongwu, 2019-09-15:
@olofk I run the stress tests on several platforms with your code, I can't see any problem so far.  could you please make sure you have CONFIG_TIMESLICING enabled? You can check it in file .config. 

And for " there is no reasonable explanation for some of the register values which makes me suspect problems during a context switch." could you please talk more about this? actually when running main thread, only interrupt will trigger context switch. I'm not familiar with your platform, we should fix this issue together. Thanks

olofk, 2019-09-15:
@wentongwu Yes, timeslicing is enabled

    grep -r TIMESLIC
    zephyr/.config:CONFIG_TIMESLICING=y
    zephyr/.config:CONFIG_TIMESLICE_SIZE=0
    zephyr/.config:CONFIG_TIMESLICE_PRIORITY=0

Also, I have not been able to reproduce the issue in a simulator. Apart from any potential synthesis issues, the main thing that differs between the simulator and the real HW is that the memory latency is much lower in the simulator. Not sure if that would make any difference in this case though.

What I mean with the unreasonable values is that just before the crash, some register value is being set from immediate constants, but there is another value showing up when Zephyr dumps the register values after the exception. A sequence could for example look like this

    lui a5,0x5 #a5 is now 0x5000
    addi a5,a5,-720 # a5 is now 0x5000-720=0x4d30
    lw a2,0(a5) # <= Crash here because a5 contains garbage which triggers an unaligned load exception

Because of that I suspect that the PC have just returned after serving a timer interrupt (which is the only interrupt source in the system) to execute the third instruction and a5 has for some reason been overwritten.

Since I don't really understand the scheduler code I find it a bit hard to know what to look for. The commit message of the original fix seems to indicate that this fix should avoid waking up if there's nothing to do. Is there something I should look at to verify that things are working as they are supposed to?

In the screenshot below I am running the code in a simulator. I'm not really sure this is what it's supposed to look like. When the p* signals are toggling, that's when we are writing to the UART, and I suppose this is where the thread is doing actual work, while there are long periods where we get a timer interrupt and the timer driver sets the new timeout value to the same as the old one, causing a new interrupt to immediately trigger. Perhaps this is not a scheduler issue at all, but some bug/configuration error in the timer driver that somehow gets exposed by this change.

[image]

wentongwu, 2019-09-18:
@olofk AFAIK, if CONFIG_TIMESLICE_SIZE=0, the time slicing will not be started. The timer driver developed by yourself or just use the existing one? Do you run the test cases located in tests/kernel before/after this change?

andyross, 2019-09-18:
Throwing some more questions on the table:

* It's not an existing test, it's your own code?  How many threads are runnable at the time of the crash, and what is their priority values?

* Reverting just that one change, and nothing else, fixes the problem, correct?

* Does setting CONFIG_TIMESLICING=n also work?  How about setting CONFIG_TIMESLICE_SIZE to some non-zero value?

I'm as confused as @wentongwu, it certainly seems that, for the configuration you posted, that the fix in question should be a noop.

olofk, 2019-09-19:
Just wanted to let you know that I'm eager to get this resolved but I'm fighting against a deadline so I won't be able to work on this for another week.

dleach02, 2019-10-08:
@olofk, can you provide some feedback to @andyross questions?

olofk, 2019-10-08:
@dleach02 @andyross My apologies, but I have not yet had time to do additional testing. But to answer some of the questions:

This is the Zephyr hello world sample with the only change being that I run the printf it in a while(1) loop and print out an increasing index. I don't know how to check for the number of threads running but I haven't explicitly started any, and the only drivers enabled are the UART (in polling mode) and timer so I guess there shouldn't be any other threads running

Yes, only reverting that change will fix the issue

I haven't had time to test other timeslicing options yet. Will let you know when I have had time to do that

olofk, 2019-10-10:
I have done further investigations now and this is not a Zephyr issue but a hardware issue that started appearing with a lot higher frequency after this change. Closing this as there is nothing to fix on the Zephyr side

closed 2019-10-10 by olofk