zephyrproject-rtos/zephyr #19134: Add 3-wires support to SPI protocol

avisconti, 2019-09-13:
Some chips provides SPI with a half-duplex single data line (MISO and MOSI shares the same wire). Interfaces like this are commonly called "3-wire SPI".
It seems that zephyr is missing the SPI 3-wires mode feature.

I think that "struct spi_config" should be enhanced to provide the possibility to pass SPI-3-WIRES-MODE. Then drivers would accept or discard the option according to their capabilities.

avisconti, 2019-09-16:
@galak @erwango @pabigot 
I put you too here as I feel it is something we should have in zephyr.

chris-makaio, 2019-09-19:
@avisonti just came across this issue because I'm working with a device which only supports 3-wire SPI. At least for nRF52840 it seems to work out of the box - I see no problems having MISO set but not in use

avisconti, 2019-09-26:
> @avisonti just came across this issue because I'm working with a device which only supports 3-wire SPI. At least for nRF52840 it seems to work out of the box - I see no problems having MISO set but not in use

@chris-makaio 
Thank you for the info. Currently I stopped investigating the SPI 3-wire because I discovered that on the board I'm working on (SensorTile.box) I can use SPI 4 wire for all the sensors. 
Anyway, are you saying that you didn't do anything special to make SPI working in 3-wire mode?

chris-makaio, 2019-09-26:
Correct!

pabigot, 2019-09-26:
@chris-makaio I'm a little confused about that.  You say you have "MISO set but not in use"; that suggests you're not using a half-duplex true 3-wire SPI (where MOSI is bidirectional) but a SPI device that doesn't use MISO (such as a common write-only OLED display).  That's a unidirectional 3-wire SPI interface, not what's being requested here.

According to https://devzone.nordicsemi.com/f/nordic-q-a/9750/3-wire-interface-on-nrf52 and other answers on that site Nordic doesn't support true 3-wire SPI. You can fake it by reconfiguring the peripheral by changing whether the pin is assigned to MISO or MOSI between each read and write transition, or by shorting MISO and MOSI through a 10k resistor.

In any case Zephyr doesn't seem to have any support to configure for (bidirectional) 3-wire SPI so for the general case this seems to be a true functionality gap.

chris-makaio, 2019-09-26:
Oh sorry, I simply thought that 3-wire SPI meant half-duplex with no MISO data, shame

avisconti, 2019-09-27:
> @chris-makaio I'm a little confused about that. You say you have "MISO set but not in use"; that suggests you're not using a half-duplex true 3-wire SPI (where ~MOSI~ SDIO is bidirectional) but a SPI device that doesn't use MISO (such as a common write-only OLED display). That's a unidirectional 3-wire SPI interface, not what's being requested here.

@pabigot 
Ah ok. Yes, that's why it is working in his case.
Thanks for pointing it out!

erwango, 2019-09-30:
@avisconti, going quickly through the comments there is no more need for this ticket, am i right ?

pabigot, 2019-09-30:
@erwango That depends on what "need" means:

> In any case Zephyr doesn't seem to have any support to configure for (bidirectional) 3-wire SPI so for the general case this seems to be a true functionality gap.

Zephyr doesn't support 3-wire SPI, so we need the issue.  The original requester no longer has case where 3-wire SPI is required, so we don't need the issue.

I think it should remain open at low priority.

avisconti, 2019-09-30:
> @erwango That depends on what "need" means:
> 
> > In any case Zephyr doesn't seem to have any support to configure for (bidirectional) 3-wire SPI so for the general case this seems to be a true functionality gap.
> 
> Zephyr doesn't support 3-wire SPI, so we need the issue. The original requester no longer has a case where 3-wire SPI is required, so we don't need the issue.
> 
> I think it should remain open at low priority.

Yes, I confirm.
I think that would be really nice to add 3-wire support to SPI API sooner or later.
But yes, I am not requiring it as for what concern sensortile.box board support.

tbursztyka, 2021-07-07:
I'll look into this in a few weeks  (I need 3-wire mode supported to hook some sensors at home so...)

commit f8cc93de (2021-11-19): spi: Add a configuration option for half/full duplex mode

This is meant to expose half/full duplex configuration option on regular
configuration. So far, dual/quad/octal are not exactly supported, as it
would require extensions to the SPI buffer for a full support.

So moving these modes to an extended operation attribute
(32 vs 16 bits), disabled by default.

And exposing half/full duplex configuration bit. Full duplex being the
default option.

Fixes #19134
closed 2021-11-19 by carlescufi