zephyrproject-rtos/zephyr #19152: MK22F51212 MPU defines missing

bfrog, 2019-09-13:
kinetis MPU defines are missing for MK22F51212 which means the MPU code will fail on SYSMPU being undefined

```
 tburdick@jupiter  ~/zephyrtrees/modules/hal/nxp/mcux/devices   master  rg "#define SYSMPU " MK64F12                                                                                                                                                                                               ✔  10131  14:08:05 
MK64F12/MK64F12.h
20049:#define SYSMPU                                   ((SYSMPU_Type *)SYSMPU_BASE)
 tburdick@jupiter  ~/zephyrtrees/modules/hal/nxp/mcux/devices   master  rg "#define SYSMPU " MK22F51212                                                                                                                                                                                            ✔  10132  14:08:12 
 tburdick@jupiter  ~/zephyrtrees/modules/hal/nxp/mcux/devices   master     
```

agansari, 2019-09-16:
@bfrog K22F has no memory protection, check manual

bfrog, 2019-09-16:
@agansari that makes the most sense actually, this post in particular confused me making me think perhaps just the software support was missing for it https://community.nxp.com/thread/499985

DerekSnell, 2022-11-17:
I can confirm, the MK22F51212 device has no MPU.  @teburd , can we close this issue?

closed 2022-11-17 by teburd