zephyrproject-rtos/zephyr #19164: compiling posix 64 with unistd.h & net/net_ip.h fail

ehudmarvell, 2019-09-15:
posix64 comilation fail on:

> /usr/include/unistd.h:277:21: error: conflicting types for ‘socklen_t’
>  typedef __socklen_t socklen_t;
>                      ^~~~~~~~~
> In file included from ../src/main.c:9:0:
> /home/ehudn/work/new_zephyr/zephyr/include/net/net_ip.h:154:16: note: previous declaration of ‘socklen_t’ was here
>  typedef size_t socklen_t;
>                 ^~~~~~~~~

**To Reproduce**
1.
add to helloworld app the following lines

> #include <net/net_ip.h>
> #include <unistd.h>

2. cd samples/hello_world/
3. mkdir build
4. cd build
5. cmake -GNinja -DBOARD=native_posix64 ..
6. use the attaching config
[config.txt](https://github.com/zephyrproject-rtos/zephyr/files/3613779/config.txt)
7. ninja
8. See error

**Expected behavior**
compilation pass

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment (please complete the following information):**
 - OS: LINUX
 - Toolchain Zephyr SDK
 - Commit SHA: a4f4f6d6f3c580e9230ec0b3de567bb786b3319e

pfalcon, 2019-10-23:
native_posix* targets: a) leak host headers into the build; b) not compatible with Zephyr POSIX subsystem (because there's no proper separation between host vs Zephyr POSIX layers). This is known problem, see e.g. https://github.com/zephyrproject-rtos/zephyr/issues/13054

Unassigning from myself and assigning to @aescolar. I'd also suggest to lower priority of this (because this is effectively a known limitation of native_posix* targets).

aescolar, 2019-10-23:
@ehudmarvell This is a not a bug as such, as much as something that was willingly chosen to stop supporting :( . You can see some history in #13054. You can also see #13540 where the [POSIX API shim was disabled in native_posix targets](https://github.com/zephyrproject-rtos/zephyr/pull/13540/files#diff-e9002f99f8723c1e1e2492da615d0e88).
I will be closing this ticket as you are trying to do something which is just not supported. If you would like this to be supported you could either create a "feature request", or chip-in in #13540

closed 2019-10-23 by aescolar