zephyrproject-rtos/zephyr #19167: Message queues bug when using C++

jcyrax, 2019-09-15:
When using message queues API with C++ and CONFIG_SMP and SPIN_VALIDATE are both disabled, there are bugs with k_msgq structure member alignment. Message queue API inline functions will return garbage values.

As far as I understand this is because C and C++ have different rules for empty structures.
k_msgq has struct k_spinlock as nested structure, but if CONFIG_SMP and SPIN_VALIDATE are both disabled, k_spinlock has no members and C treats it like zero size object, but C++ treats it as 1 byte size object. So it happens that k_msgq_init inits some members at wrong offset (from point of view of C++), but code that uses API inline functions from kernel.h with C++ compiler assumes different alignment. So functions like k_msgq_num_free_get will return wrong values if used from unit compiled with C++.

**Environment:**
 - Linux
 - arm embedded toolchain: gcc-arm-none-eabi-8-2019-q3-update
 - zephyr: aa066d413945a4bfcb9b36963a859e6bc7de4536

lowlander, 2019-09-15:
@jcyrax  I added a PR, maybe you can give it a try #19168

lowlander, 2019-09-16:
A fix should now be in master

closed 2019-09-16 by jcyrax