zephyrproject-rtos/zephyr #19177: re-valuate commit 0951ce2

avisconti, 2019-09-16:
Commit 0951ce2 is introducing the possibility to disable interrupt at runtime for STM32.
Nevertheless, since the EXTI lines are shared among group of pins (i.e. PA0-PI0 shares EXTI0, PA1-PI1 shares EXTI and so on), disabling EXTI0 means disabling interrupts for all group 0 of pins (e.g. Pa0/PB0/PC0/...). THis is wrong in my opinion.

As an example, in my sensortile-box bsp (see #18182) I'm using PA2 and PD15 as sensor interrupts and PB15 and PF2 as led output. When configuring PB15 and PF2, since they are not GPIO_INT, the exti is disabled also for PD15 and PA2 (group 15 and group 2 are exti disabled).

I would like to seek for the help of @erwango  (I assigned the bug to him) and @mniestroj who is the author of the commit.

erwango, 2019-09-16:
Thanks @avisconti, looking into it.

erwango, 2019-09-16:
@avisconti, @mniestroj, will push a revert fixup as this could indeed be a source of regression in many cases. Then we'll take time to discuss the point.

avisconti, 2019-09-16:
> @avisconti, @mniestroj, will push a revert fixup as this could indeed be a source of regression in many cases. Then we'll take time to discuss the point.

@erwango 
agree.
I think the intention of the commit would require some time of re-thinking and new proposals.

erwango, 2019-09-16:
@mniestroj, can you elaborate why using existing gpio api function `gpio_remove_callback` was not fitting your need ?
(I should have start with that from the beginning)

closed 2019-09-16 by galak

mniestroj, 2019-09-18:
@erwango `gpio_remove_callback` only disables application callback, but interrupts are triggered anyway. If noone is interested in the interrupts anymore, then there is no sense to keep them enabled in the MCU. Disabling them in MCU should also result in lower power consumption, because: 1) CPU does not have to clear interrupt flags after each interrupt is triggered, 2) EXTI does not have to keep the edge detection logic enabled (if that consumes any power at all).

So keeping in mind that on STM32 CPUs we are only able to have single GPIO port enabled for single interrupt line, then we can simply check whether the configured GPIO port is the one that we are configuring right now:
```
		if (gpio_stm32_int_enabled_port(pin) == cfg->port) {
			stm32_exti_disable(pin);
			stm32_exti_unset_callback(pin);
		}
```

What do you think about that?

erwango, 2019-09-18:
@mniestroj , I get your point.
Can you push a new PR and we'll discuss on this proposal (and get it tested more thoroughly),will be easier than discussing in this closed thread.

PR #19179 (2019-09-16): Revert "gpio: stm32: support disabling and reenabling interrupts on pin"
This reverts commit 0951ce2d34b6b8d9966d8a08161e910c48bd02a4.

This change introduces regression on GPIO interrupts handling
when several GPIOs are configured.

Fixes #19177
commit c9dcf37e (2019-09-16): Revert "gpio: stm32: support disabling and reenabling interrupts on pin"

This reverts commit 0951ce2d34b6b8d9966d8a08161e910c48bd02a4.

This change introduces regression on GPIO interrupts handling
when several GPIOs are configured.

Fixes #19177